esp-rs/esp-idf #217: Pindriver drop should not reset pin

dberlin, 2023-03-15:
Currently, drop of a pindriver resets the pin to disabled, and destroys all subscriptions.
It is not obvious why it does this, but it has a lot of bad effects:

1. temporary pindrivers to set levels not only don't work, they actually disable the pin.

IE calling PinDriver::output(periphs.pins.gpio11).set_high() not only does not set the pin high, it will disable the pin and destroy any other interrupt handlers that exist for that pin

This is really unexpected behavior - no other Rust HAL works like this.
They all leave things as they are.  This is true of STM32, RP, NRF52840, etc.  They all do the same thing. Only ESP is different.

2. changing input/output mode of the pin doesn't just change the input/output, it resets the pin entirely, and destroys all subscriberrs.
That is, taking an existing PinDriver and calling into_input on it resets the pin entirely, rather than just change the mode.

It also causes complicated logic where into_mode tries to drop the old value before setting the new one to avoid the pin not changing mode at all.  If the drop was not here, the pin would actually never change mode at all.

3. Because of this,  Pin drivers must be created only once and Arc'd or very very carefully managed - otherwise the pin will reset in unexpected places.

None of this is really necessary, but really detrimental.  All other rust HAL's  i'm familiar with are happy to have people create and destroy and whatever the equivalent of their pin drivers, with no ill effects.
None of them implement or reset the pin on drop.

I would argue very strongly the behavior here should be the same - do nothing special when a pindriver is destroyed.

I'm happy to write and test patches to do it.

MabezDev, 2023-03-20:
Thanks for bringing this up! I agree, would you mind opening a PR to remove the drop behaviour? It's probably a good idea to add a public reset method for those that really want to disable the pin too.

dberlin, 2023-03-20:
Will do!

ivmarkov, 2023-03-22:
I would like to see at least one link to all other drivers that **don't do this** as I vaguely recall the same discussion on the embassy forums where a similar behavior was discussed.

Also: not unsubscribing from interrupts when the driver is dropped is in fact much worse behavior don't you think? If you want to keep the subscriptions running, you **must** hold on to the driver forever. And this does not mean arc-ing it.

ivmarkov, 2023-03-22:
[This](https://github.com/embassy-rs/embassy/blob/master/embassy-nrf/src/gpio.rs#L338) looks like a reset-on-drop to me, or am I misunderstanding something?

dberlin, 2023-03-22:
(The subscription is dropped, but pin is not reset, specifically to address the issue you cite earlier)

I did not look at embassy as it is bare metal, and i would expect esp-hal to mirror it, not esp-idf-hal. I expect esp-idf-hal to be those more close to coming from Arduino or other platforms, not embassy.

I am happy to look through it, but at that point you have to decide who you are targeting - i would argue, again pretty strongly, the behavior above is going to be wildly surprising to those coming from non-rust platforms.
I expect the bare metal people to always look closely at what anything does.

Meanwhile, here is the survey of non-embassy HAL's

here is the stm32 hal:
https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/gpio.rs

the only time it resets mode on drop is if you use the with_mode call, which is meant to temporarily change the mode.
IE x.with_mode(Input).something.
In that case, it will reset to previous mode (not disabled) when dropped.

https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/gpio/convert.rs#L260

On normal drop, nothing happens.

Here is NRF hal:
https://github.com/nrf-rs/nrf-hal

Nothing is done on drop for GPIO pins:
```

❯ cd nrf-hal
❯ rg -i drop
nrf-hal-common/src/i2s.rs
708:impl<B> Drop for Transfer<B> {
709:    fn drop(&mut self) {
743:impl<TxB, RxB> Drop for TransferFullDuplex<TxB, RxB> {
744:    fn drop(&mut self) {

nrf-hal-common/src/uarte.rs
620:impl<T> Drop for UarteTx<T>
624:    fn drop(&mut self) {
643:impl<T> Drop for UarteRx<T>
647:    fn drop(&mut self) {

nrf-hal-common/src/twis.rs
535:impl<T: Instance, B> Drop for Transfer<T, B> {
536:    fn drop(&mut self) {

nrf-hal-common/src/spis.rs
471:impl<T: Instance, B> Drop for Transfer<T, B> {
472:    fn drop(&mut self) {
516:impl<T: Instance, TxB, RxB> Drop for TransferSplit<T, TxB, RxB> {
517:    fn drop(&mut self) {

```

Nothing happens on GPIO drop.

Here is rp-hal:
https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/src/gpio/pin.rs

```
rp2040-hal/src/multicore.rs
37:use core::mem::ManuallyDrop;
158:                entry: &mut ManuallyDrop<F>,
163:                let entry = unsafe { ManuallyDrop::take(entry) };
186:            // We don't want to drop this, since it's getting moved to the other core.
187:            let mut entry = ManuallyDrop::new(entry);
197:                stack_ptr.cast::<&mut ManuallyDrop<F>>().write(&mut entry);
239:                        // so we have to drop it ourselves.
240:                        drop(ManuallyDrop::into_inner(entry));

rp2040-hal/src/sio.rs
410:///     // The lock is dropped here.
477:impl<const N: usize> Drop for Spinlock<N>
481:    fn drop(&mut self) {

rp2040-hal/src/gpio/pin.rs
501:        // Safe because we drop the existing Pin
1242:                /// dropped.

rp2040-hal/src/dma/single_buffer.rs
63:// TODO: Drop for most of these structs

rp2040-hal/src/spi.rs
328:                    // drop read wordd
348:                    // read one word. Drop extra words if buffer is full.

rp2040-hal/src/timer.rs
369:        impl Drop for $name {
370:            fn drop(&mut self) {

rp2040-hal/src/pio.rs
1211:impl<'sm, SM: ValidStateMachine> Drop for Synchronize<'sm, SM> {
1212:    fn drop(&mut self) {

rp2040-hal/src/pwm/mod.rs
278:        // Safe because we drop the existing slice

on-target-tests/Cargo.toml
32:name = "dma_drop"

on-target-tests/tests/dma_drop.rs
85:    fn droptest_runfirst(state: &mut State) {
89:            .expect("Failed to take Channels, not dropped correctly");
104:    fn droptest_runsecond(state: &mut State) {
108:            .expect("Failed to take Channels, not dropped correctly");
```

The full context of the drop in gpio is this:

```
    /// Convert the pin to the requested [`PinMode`]
    #[inline]
    pub fn into_mode<N: PinMode + ValidPinMode<I>>(mut self) -> Pin<I, N> {
        if N::DYN != M::DYN {
            self.regs.change_mode::<N>();
        }
        // Safe because we drop the existing Pin
        unsafe { Pin::new() }
    }
```

IE on mode change it expects the old pin structure to disappear

It does not reset any pins on drop.

I can go through more if you want, but they are consistent with this.

ivmarkov, 2023-03-22:
I'm not changing the current behavior as I'm currently not finding your arguments convincing enough, sorry. Perhaps, if you look deeper as to how `esp-idf-hal`, `esp-hal` and the embassy drivers work, you'll get convinced too. Also - some details in that direction, as well as addressing some of your arguments in concrete below.

> I did not look at embassy as it is bare metal, and i would expect esp-hal to mirror it, not esp-idf-hal. I expect esp-idf-hal to be those more close to coming from Arduino or other platforms, not embassy.

No. It does not matter if there is a hidden piece of C code underneath (`esp-idf-hal`) or not (`esp-hal`). Both are supposed to follow Rust idioms here and provide a type-safe Rust API, not C idioms. Your comparison does not hold IMO. The reason why the `esp-idf-*` crates exist is not for folks coming from C (or Arduino-C++) to be able to re-use their language skills in Rust somehow, but to provide wrappers around unsafe but well tested drivers (ESP IDF).
 
Shall I even mention, that you ARE however looking at the other (older) bare-metal HALs though, which is kind of strange. I mean... they are bare metal too, right? Why comparing to `esp-idf-hal`, according to your own logic?... :)

> I am happy to look through it, but at that point you have to decide who you are targeting - i would argue, again pretty strongly, the behavior above is going to be wildly surprising to those coming from non-rust platforms. I expect the bare metal people to always look closely at what anything does.

I am targeting:
* Similar behavior to the embassy crates, which for me are a more modern and more ergonomic evolution of the older Rust HAL crates. Did you actually notice, that the crates you are citing don't even have the notion of a "driver" w.r.t. GPIO and instead heavily rely on typestates by fusing the driver and the peripheral into one thing; contrary to both the embassy trates (`Flex` and `Input`) as well as `esp-idf-hal` (`PinDriver`) and hopefully soon `esp-hal` will follow suit. Did you actually also notice the notion of `PeripheralRef` which the old drivers don't have, contrary to `esp-idf-hal` and embassy (from where I copied it) and`esp-hal` followed later? Do you understand that the old Rust HALs in fact *don't* have destructors, as they are still practicing the old less ergonomic `release()` pattern, rather than `PeripheralRef`?
* Similar behavior between `esp-idf-hal` and `esp-hal`

I'm specifically NOT targeting at all:
* Similar behavior between `esp-idf-hal` and the Arduinos of this world, for the reasons above.

As for "nothing happens" on drop with the old drivers - sure - `drop` is useless there as if you drop the driver, the peripheral is gone with the wind too, as they are unable to borrow the peripheral (which the `PeripheralRef` magic from above does); they still practice `release`/`free` pattern instead which I removed ~ 6months ago. And again and on top of that - they are also fusing the notion of GPIO peripheral and GPIO driver - unlike all other drivers (i2c, spi, uart, etc), which is awkward.

ivmarkov, 2023-03-22:
And for the need to "arc" the driver - if you are willing so much to configure the pin and then leave it configured without actually using it - which is kinda awkward but oh well... just `core::mem::forget` the driver.

dberlin, 2023-03-22:
We are not going to agree on this, clearly.
I gave you some examples, which you simply dismissed out of hand after asking for them.  Okay.
I can give you non-bare metal hals if you want in the end, but honestly, based on your tone and approach, i am left with zero doubt whether it will matter or not.    I'm not sure any amount of evidence will convince you, because it's totally unclear what actual *data about your users* you are basing your decision on, which is really what should matter.  Technical rightness is fun but uninteresting if your goal is actually to build something users want to use.
Instead, it seems you have a very strong notion of what you think is right, and want to follow that - have at it.

I'm very familiar with both Rust and C idioms, and others, actually and honestly, don't find your arguments convincing.  I simply pointed out one user base that would be seriously confused. I can point out others.
But again,  it's your show.

In the end, all i'm telling you is that as a user, this behavior is incredibly confusing and annoying, you can do with it what you want.

Honestly, given my experience not with your decisions, but how the project takes feedback - everything is met with super strong opinions rather than attempts to understand, i'm gonna go do something else with my time.

I wish you the best of luck building an ecosystem people want to use.

closed 2023-03-22 by dberlin

FrigoEU, 2024-01-29:
Fwiw, this was extremely annoying as a user new to this library / ecosystem. Not in a milion years did I expect the current behavior. I would much prefer the behavior described by @dberlin.