esp-rs/esp-idf #228: Multi-core tasks and queues

wouterdebie, 2023-03-31:
I'm tempted to port an ESP-IDF project of mine to rust using rust-idf-hal and rust-idf-svc, but one thing that I can't seem to figure out is how to use multiple cores. In my C code I heavily rely on creating tasks (`xTaskCreate()`, `xTaskCreatePinnedToCore()`) and communicating between tasks using a bunch of the FreeRTOS synchronization mechanisms like `QueueHandle_t`, `xQueueReceive()` and `xQueueSend()`. I haven't been able to find equivalents in the rust esp world, except for unsafe functions in `esp-idf-sys`. 
What is considered an idiomatic way of dealing with multiple cores and communication between them? Is it rust threads and rust's own synchronization mechanisms?

Vollbrecht, 2023-03-31:
Did you had a chance to look at the https://github.com/ivmarkov/rust-esp32-std-demo ? a thread based approach works absolutely fine. with the FreeRtos support of pthreads, we have a thin wrapper around the xTaskCreate mechanism. Have a look at esp-idf-hal/tasks.rs. you can use the rust std::thread builder pattern on top of it. Short example -> 
```
    ThreadSpawnConfiguration {
        name: Some("mid-lvl-thread".as_bytes()),
        stack_size: 10000,
        priority: 15,
        ..Default::default()
    };

    let _thread_1 = std::thread::Builder::new()
        .spawn(move || {
               //do stuff
        })
        .unwrap();

```
we have the eventloop system exposed via safe wrappers. you can also rely on other rust std mechanisms for synchronization

Vollbrecht, 2023-03-31:
for an notification system wrapped around xTaskCreatePinnedToCore you can look @ esd-idf-svc/notify.rs .  You can also try using std::sync::mpsc etc. Or using our approch of the async edge-executor running async codes in one or different threads and using its synchronization methods around ( embassy etc)

wouterdebie, 2023-03-31:
@Vollbrecht this is super helpful! Thank you! Would it be helpful to have functionality around FreeRTOS queues in  `esp-idf-svc`? Something like esd-edf-svc/queues.rs? I might think about contributing something like this to the project in if it's helpful.

Vollbrecht, 2023-03-31:
If you think this is useful for you there is a chance that can be useful for others. So PR's are always welcome. White that said, the question becomes. If all the code you are building now are build completely in rust, it bags the question if its maybe more useful to use the std primitive here rather than calling into the C layer. ( Maybe its better maybe its not, maybe you can build something and try to find out for comparison). In the rust world there are more alternatives to sync::mpsc, thinks like crossbeam-channels or flume. 
But honestly i can not give a good answer here, i currently lack experience in this region with using this on an esp. To much to test out and this is still a blind spot for me.
@ivmarkov could properly answer it better if this would be an useful addition.

wouterdebie, 2023-03-31:
I think you're right. It seems that sync::mpsc does pretty much the same thing as the queues in FreeRTOS, so there doesn't seem to be a good reason for a reimplementation. I think I'll fiddle with threads and mpsc a bit to get a bit more familiar with it. Again, thanks for the insights!

closed 2023-04-02 by wouterdebie

oguzemrec, 2023-08-10:
Can someone explain how ThreadSpawnConfiguration works? I have been unable to find any examples that demonstrate how it works. I have also tried using it on an ESP32 chip, but I have not observed any changes.

svenstaro, 2023-08-12:
@oguzemrec It influences how the next threads are spawned. It's a bit awkward to use:
```rust
ThreadSpawnConfiguration {
    name: Some(b"hello\0"),
    ..Default::default()
}
.set()
.unwrap();

thread::spawn(|| {})

// Set it back to defaults.
ThreadSpawnConfiguration::default().set().unwrap();
```

oguzemrec, 2023-08-19:
Thanks for the info @svenstaro.

smallbraingames, 2024-10-26:
Where can I find / learn about how the wrapping code around `xTaskCreate` is generated? (looking to spawn tasks, etc. in a no-std context)

ivmarkov, 2024-10-26:
> Where can I find / learn about how the wrapping code around `xTaskCreate` is generated? (looking to spawn tasks, etc. in a no-std context)

In the ESP-IDF FreeRTOS documentation.

With that said, using `no_std` with `esp-idf-*` is a very niche case that you almost certainly don't need.

If you really want to use `no_std` you should probably try `esp-hal`.

smallbraingames, 2024-10-27:
> In the ESP-IDF FreeRTOS documentation.
>
> With that said, using `no_std` with `esp-idf-*` is a very niche case that you almost certainly don't need.
>
> If you really want to use `no_std` you should [...]

ok got it

I just barely use any `std` features (only threading), so figured I could build without std and call those directly. reduces binary size

Tried `esp-hal` but found it unstable (notably, dhcp in wifi didn't work) so moved to the idf wrapper

ivmarkov, 2024-10-27:
> reduces binary size
 
Are you sure about that?
Rust and linkers in general dont't quite work like that. It is not like you use one function from STD and you have the whole standard library pulled in.

But if you have evidence suggesting otherwise, please share!

> Tried `esp-hal` but found it unstable (notably, dhcp in wifi didn't work) so moved to the idf wrapper

I don't think DCHP in wifi doesn't work or else there would be an army of people complaining. Perhaps you are not using in exactly the right way?

smallbraingames, 2024-10-27:
> Are you sure about that? Rust and linkers in general dont't quite work like that. It is not like you use one function from STD and you have the whole standard library pulled in.
>
> But if you have [...]

I was likely doing something wrong, dhcp was hanging for me–but I see 2 weeks ago a new version of the lib was released with a ton of updates

I am using `esp-sr`'s acoustic echo cancellation as well (adding it as a component) so I likely cannot use `esp-hal` either way

Sticking with std because too many deps need it. I remember running the wifi example with / without std on the esp32 and w/ std having too large a binary to flash without changing partition, but without running under the limit. let me try to replicate that and get back

ivmarkov, 2024-10-27:
Going `no_std` means not just dealing with threads by yourself (look at `esp_idf_svc::hal::task` btw for a lightweight way to create native tasks for ESP IDF), but also means dealing with socket IO yourself. And this really gets ugly. As in, raw C calls to the BSD sockets API, which is not exactly beautiful.

A quick fix which often trims a lot of fatness is to just add to `.cargo/config.toml` the `build-std-features=panic_immediate_abort` setting.

smallbraingames, 2024-10-27:
> `build-std-features=panic_immediate_abort`

Wow thank you, that reduced my bin from 1536832B to 1190240B

ivmarkov, 2024-10-27:
This is basically trimming all strings that are only referenced from `panic!`s. 

You lose the message when your firmware will panic, but of course you still have the panic ESP IDF backtrace, which you can grab (somehow). I need this too and need to check how to grab it.

Trimming more strings might still be possible by replacing the `log` crate with `defmt`, but that would require laborious (though trivial and repetitive changes in the `esp-idf-*` crates themselves, where depending on whether `log` or `defmt` was selected, they log either via log, or via defmt.

... and implementing the defmt equivalent of `Debug` instead of the usual `Debug`. Not sure if there is a defmt equivalent of `Display`. Should be.

ivmarkov, 2024-10-27:
Ah, it is another topic that ESP IDF itself of course does not know anything about `defmt`. Oh well. Maybe not so trivial.

smallbraingames, 2024-10-27:
> This is basically trimming all strings that are only referenced from `panic!`s.
>
> You lose the message when your firmware will panic, but of course you still have the panic ESP IDF backtrace, which [...]

something like this https://lib.rs/crates/defmt-or-log perhaps?

and right...