esp-rs/esp-idf #230: esp-idf-hal will support the embedded-hal timer traits?

jmmb13, 2023-04-01:
Hello,

Does esp-idf-hal is planning to support embedded-hal::timer traits in TImerDriver?

Thank you!
jmmb

Vollbrecht, 2023-04-01:
I think they are no current plan's for it. This is manly because the embedded-hal timer was axed in the current itterations of the embedded-hal 1.0.0.alpha releases. So i think we don't want to add api's that currently are already deprecated. If they are plan's in the 1.0.0.alpha release or on master branch for timer specific traits, we will support such things for sure. Its a though position with the alpha traits currently.

Vollbrecht, 2023-04-01:
for more info and tracking look here: https://github.com/rust-embedded/embedded-hal/issues/359

closed 2023-05-13 by ivmarkov