esp-rs/esp-idf #231: Does this crate support any USB function?

windoze, 2023-04-03:
I can see the USB support exists in `esp-hal`, but I failed to find anything about USB in this crate.

Is it possible to use USB with this crate, such as ESP-IDF builtin TinyUSB components, or something equivalent to `Peripherals::USB0`?

Vollbrecht, 2023-04-03:
I think there is currently no wrapper written for TinyUSB in this crate. This is only effecting esp32s2/s3 and because other's don't have USB CDC-ACM support correct? Currently one would have to use esp-idf-sys and interact with it directly, though i am not sure if all the headers are exposed. When using esp-idf v4.4 it should be included, but with v5 it is now an external component. 
So to answer your question -> I am not sure what it is in no_std about Peripherals::USB0, but you can use the normal USB functionality to use it as an virtual serial connection, or use it for jtag debugging and programming. For further TinyUSB support - PR's are always welcome. But thanks for bringing it up on the radar.

windoze, 2023-04-03:
Thanks for the reply.
I tried several approaches to include TinyUSB into the project which uses `esp-idf-hal` but didn't work.
I don't have too much experience with ESP-IDF, could you please show me some clues or hints about the task?

Vollbrecht, 2023-04-03:
If you don't have a lot of experience with ESP-IDF itself than this is a hard problem. Because you need to directly interact with the provided c functions from ESP-IDF to get it started. If you want to give it a shot here are some rough pointers.
Assuming you are building ontop of esp-idf v4.4.4.
 1.  read the TinyUSB esp-idf docs extensively and understand what the complete API can do [DOCS](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32s3/api-reference/peripherals/usb_device.html)
 2. make sure you can interact with the API through esp-idf-sys -> Some simple way to checkout if the headers are exposed to rust is to have a look inside the bindings.rs generated on build. Build a hello world program with esp-idf-sys and find it somewhere in target/xtensa.../debug/build/esp-idf-sys.../out/bindgen.rs 
 3. Interact with the esp-idf API through it.
 4. Profit
 
Depending on your knowledge this is a fun project, but you should have a good grasp of thinks first. If you want to learn more best is to try to understand the wrapper code we already written and how are things can be done. Have fun with this info.

windoze, 2023-04-04:
Thanks for the guidance.

Finally, I managed to make every piece work together, an demo project is at https://github.com/windoze/esparrier-idf, which is an ESP32-based USB HID device with a Barrier/Synergy client to allow the user to use remote keyboard and mouse on the local computer without needs to install any driver or app.
Despite a lot of hacks, the program works and the only existing problem is that ESP32S3 seems to have huge lags (sometimes 100+ms) over WIFI, but I think it's totally off-topic here.

Thanks again for the information and your kind help.

windoze, 2023-04-04:
One more question, does this crate support ESP-IDF 5? I can only find settings for 4.4 and master.

Vollbrecht, 2023-04-04:
some guys are using it with esp-idf 5. Though we not made it official a switch but it will probably happen not in the far future. The settings you are talking about are about cargo generate template ? You can edit it manually to set it for example to the taged version v5.0.1 and try it out.

windoze, 2023-04-04:
I tried to create a brand new project from `esp-idf-template` and change the ESP-IDF version to `release/v5.0` in `.cargo/config.toml`, also I changed `rustflags` to `["--cfg", "espidf_time64"]` as comments instructed.
But `cargo build` reported an error:
```
   --> /Users/windoze/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.45.0/src/espnow.rs:126:53
    |
126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
    |                                                ---- ^^^^^^^^^^^^^^^^^^^ expected struct `esp_now_recv_info`, found `u8`
    |                                                |
    |                                                arguments to this enum variant are incorrect
```

Do I need to use git head of `esp-idf-svc` and/or `esp-idf-hal` to make it work?

Vollbrecht, 2023-04-04:
First we deviate quite a bit from the original issue -> For quick questions the matrix chat is always a good place, alternative create new issues for new topics. 
That said when using first its always best to not use release/5.0 or release/v4.4 as they are moving git targets. better use the fixed release tags -> v5.0.1 or v4.4.4. Yes it is better to use git master of esp-idf-hal/svc for using esp-idf 5 currently

windoze, 2023-04-04:
Sorry for being off-topic, I'll close this one and submit new issues when I've further questions.

Thanks again for your help.

closed 2023-04-04 by windoze