esp-rs/esp-idf #233: Creating a Thread with ThreadSpawnConfiguration ignoes stack_size value

Vollbrecht, 2023-04-09:
```rust
ThreadSpawnConfiguration
        name: Some("Thread-A\0".as_bytes()),
        stack_size: 10000,
        priority: 20,
        ..Default::default()
    }
    .set()
    .unwrap();

let a = std::thread::Builder::new()
        //.stack_size(10000)
        .spawn(move || 
    {});
```
only works with the stack_size method with the thread::Builder

thomas725, 2023-07-28:
I believe I found a similar bug with the name but in the opposite direction:
If a ThreadSpawnConfiguration has been created and set any time, all threads spawned afterwards (before setting a new ThreadSpawnConfiguration instance) will carry that name. If the thread::Builder is given a different name, it is ignored. thread::Builder names are only honored for threads created before the first ThreadSpawnConfiguration has been set.

Vollbrecht, 2023-07-29:
nice observation, i think that might be the case because if i understand it correctly, the config will just be written to a special region, but we never clear it after we set it.  So consequently if you don't create a new ThreadSpawnConfiguration it doesn't override that region, but i could be wrong here.