esp-rs/esp-idf #234: ThreadSpawnConfiguration ignores priority greater than 22 and silenty sets it to 5

Vollbrecht, 2023-04-09:
```rust
ThreadSpawnConfiguration
        name: Some("Thread-A\0".as_bytes()),
        priority: 25,
        ..Default::default()
    }
    .set()
    .unwrap();

    let a = std::thread::Builder::new()
        .spawn(move || 
    {});
```
I could not find a clue about this behavior in the esp-idf docu. If someone knows the place please tell. Otherwise we maybe should emit a warning to the user if he sets a value bigger than 22.

MabezDev, 2023-04-11:
I presume you meant 25 in the title, based on your example. This priority comes from [this config option](https://github.com/espressif/esp-idf/blob/f404fe96b17692e3f1de536a3d73a180cdb53b42/components/freertos/esp_additions/arch/xtensa/include/freertos/FreeRTOSConfig_arch.h#L46), though I'm not sure where the overwrite happens.

I guess we can fix this by adding a check in the `ThreadSpawnConfiguration`'s `set` method.

Vollbrecht, 2023-04-11:
thx for the linking the code point, though the funny part is, wherever this max is enforced, it is enforced with `priority < configMAX_PRIORITIES  ` so the max prio one can set is 24 not 25 !

Vollbrecht, 2023-04-11:
i can create an PR fixing this, but i would like to wait for the discussion around the issue with exposing an &[u8] field interface as a thread name. Untrained users ( myself include :smile_cat: ) can simply forget to add a null byte here and it silently runs the buffer over. So question becomes if we should not use something like CStr::from_bytes_with_nul since it is stabilized. Either do it by our-self and don't change the user API or the user should provide a CStr. 
Do you have an opinion on that ?

MabezDev, 2023-04-11:
Now that `CStr::from_bytes_with_nul` is in core, I think we should use this for sure imo.

ivmarkov, 2023-04-15:
Fixed.

closed 2023-04-15 by ivmarkov