esp-rs/esp-idf #237: Current nightly rust compiler breakes build

Vollbrecht, 2023-04-14:
If you encounter the problem - mostly for esp32c* users - please use no nightly compiler newer than 10-04-23. If you already upgraded you can set your compiler version like this :
```
$ rustup install nightly-2023-04-10
$ rustup component add rust-src --toolchain nightly-2023-04-10-x86_64-unknown-linux-gnu #last part depending on platform 
```
```
$ rustup default nightly-2023-04-10
``` 
the default compiler option is by default overridden in your local rust-toolchain.toml file. You might need to change it to something like this
```
[toolchain]
channel = "nightly-2023-04-10"
```

ivmarkov, 2023-04-15:
Fixed.

ivmarkov, 2023-04-15:
Note that I'm nevertheless pinning this issue, as this is still valid for the released crates, until new version is released.

closed 2023-04-15 by ivmarkov