esp-rs/esp-idf #246: Request to not use the legacy esp_adc_cal driver in ESP-IDF HAL

p3zhy, 2023-05-07:
I recently came across the following message in the ESP-IDF documentation:

"esp_adc_cal component is no longer supported. New adc calibration driver is in esp_adc component. Legacy adc calibration driver has been moved into esp_adc component. To use legacy esp_adc_cal driver APIs, you should add esp_adc component to the list of component requirements in CMakeLists.txt."

I understand that the legacy `esp_adc_cal` driver has been moved to the `esp_adc` component and that it is still possible to use it by adding the `esp_adc` component as a requirement in `CMakeLists.txt`. However, I would like to request that the use of the legacy driver be discouraged in the ESP-IDF HAL going forward.

As a developer, I believe it is important to use the latest and recommended tools and drivers, and in this case, the new `esp_adc` driver provides better calibration functionality. Therefore, I would like to request that the HAL documentation be updated to reflect this change and that the use of the `esp_adc_cal` driver be discouraged.

Thank you for your attention to this matter. I look forward to your response.

ivmarkov, 2023-05-13:
This project is running mostly on my free time as well as the partial support from one of the Espressif Rust baremetal folks.
All the rest are contributions from the community. 

Writing safe wrappers for the ESP IDF 5 ADC driver is not a rocket science, yet would still take a couple of days. We gladly accept PRs therefore!

ivmarkov, 2023-09-18:
Closing as we now have a version of the oneshot driver using the new esp-idf v5 APIs, as well as an adc continuous mode driver (v5 only).

closed 2023-09-18 by ivmarkov