esp-rs/esp-idf #248: uart: Null pointer deref since 0.41

mbuesch, 2023-05-17:
I get

```
E (841) uart: uart_intr_config(746): param null
```

which I believe comes from

```
esp!(unsafe { uart_intr_config(UART::port(), IntrFlags::to_native(config.intr_flags) as _) })?;
```

which essentially converts a `0` to a `NULL` pointer.

because:

```
pub(crate) fn to_native(flags: EnumSet<Self>) -> u32 {
...
```

```
    pub fn uart_intr_config(
        uart_num: uart_port_t,
        intr_conf: *const uart_intr_config_t,
    ) -> esp_err_t;
```

Vollbrecht, 2023-05-17:
related commit https://github.com/esp-rs/esp-idf-hal/commit/ba9cc4246a18880cdcea5aea21ec3a23edd220f6

mbuesch, 2023-05-17:
I commented out the line with `uart_intr_config()` call and the uart starts to work again.

mbuesch, 2023-05-17:
And yes, it's technically not a NULL deref. But it's a hard error, which results in an unwrap() in my case, because I can't really do anything else in my user code.
But it could be an invalid pointer deref, if the user code sets some intr flags, which I don't do.

ivmarkov, 2023-05-21:
Dupe of #250

closed 2023-05-21 by ivmarkov