esp-rs/esp-idf #249: Blinky Example doesn't do anything on ESP32 C6

seanaye, 2023-05-18:
I flashed the blinky example to my esp32 c6 using GPIO pin 8 as that is the onboard LED for that board.

The program runs but the LED does nothing.

Vollbrecht, 2023-05-18:
can you elaborate a bit more here. esp32c6 support is fresh from the press and it can be that some of the pin definitions are not correctly. What board are you using? How did you setup the blinky?  With esp-idf 5.1 option? Did you compile against the custom imac target needed for c6?

seanaye, 2023-05-18:
I will put up a minimum repro. I am using esp idf ESP-IDF v5.1-dev-2036-gb04e33256d with the custom imac target as defined in esp-idf-sys. I am using the esp32-c6-WROOM-1 devkit ordered from Mouser

seanaye, 2023-05-18:
https://github.com/seanaye/esp

here is the minimum repro, my code is a bit messy since I was doing something else

Vollbrecht, 2023-05-18:
your led on pin8 on that board is an rgb led! not a normal led. you need to talk to it and cant just toggle it

seanaye, 2023-05-19:
Thanks I will try the RMT neopixel example

ivmarkov, 2023-05-21:
@seanaye Can I close this?

closed 2023-06-01 by seanaye