esp-rs/esp-idf #251: Cannot make it work with any embedded-hal drivers tested (DHT11, i.e.)

luanmm, 2023-05-22:
There something that seems incompatible between esp-idf-hal and embedded-hal specs that just can't make sensors reading work as desired.

Drivers tested:
- https://github.com/michaelbeaumont/dht-sensor
- https://github.com/kelnos/dht-embedded-rs

When it seems to be compatible with IO specs, the `EspError` struct seems to be the problem (incompatible with GpioError).

Anyone has any clue about what can be happening? There is any sample related to digital readings like this (with similar sensors, which have their own way to process signals and needs some kind of "driver")?

Thanks in advance.

Vollbrecht, 2023-05-22:
dht-embedded-rs still uses embedded-hal alpha.9 traits - esp-idf-hal implements the embedded-hal.alpha.10 traits. The Error Handling on gpio's was changed.
So you need to use an updated version of dht-embedded-rs to alpha.10. 

We either support the old 2.7 version of embedded-hal or the newest alpha version.

luanmm, 2023-05-22:
Thank you very much for the guidance, @Vollbrecht. It seems to compile now, after following your advice.

I will test again with de HW tomorrow to validate in practice.

Best regards.

luanmm, 2023-05-23:
It worked, indeed.

But I had to use "dht11" crate, with some updates in "embedded-hal" interfaces in use.

Here is the adjusted code/repo: https://github.com/luanmm/dht11-rs.

Thank you, again, for de guidance, @Vollbrecht!

closed 2023-05-23 by luanmm