esp-rs/esp-idf #257: Multiple LED Neopixel example

seanaye, 2023-06-01:
I am using the rmt neopixel example with board hooked up to WS2812B strip. The led on the board an the first LED in the strip work but I cant figure out how to turn on more than one LED.

I have looked at the reference implementation mentioned in comments.

It looks like this C struct doesn't have an equivalent in this crate
https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/examples/peripherals/rmt/led_strip/main/led_strip_main.c#LL90C15-L90C15

How can I adapt the neopixel example to arbitrary number of LEDs

ivmarkov, 2023-06-03:
That's because the example you are referring to uses additional C code (custom component) which is not part of ESP IDF itself: https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/examples/peripherals/rmt/led_strip/CMakeLists.txt#L5

closed 2023-06-03 by ivmarkov

seanaye, 2023-06-06:
@ivmarkov I am still struggling to write to multiple LEDs, that C code example is using `rmt_write_sample` which is a deprecated API. It would be great if you could add to the examples the recommended way of writing to a strip of LEDs as this a common usecase.