esp-rs/esp-idf #264: Implementation of `InputPin` and `OutputPin` (from `embedded-hal` v0.2) for `PinDriver` have different error types

leungyauming, 2023-06-13:
For the implementation of `InputPin`, `InputPin::Error` is `EspError` as shown:
```rs
impl<'d, T: Pin, MODE> embedded_hal_0_2::digital::v2::InputPin for PinDriver<'d, T, MODE>
where
    MODE: InputMode,
{
    type Error = EspError;

    fn is_high(&self) -> Result<bool, Self::Error> {
        Ok(PinDriver::is_high(self))
    }

    fn is_low(&self) -> Result<bool, Self::Error> {
        Ok(PinDriver::is_low(self))
    }
}
```

For the implementation of `OutputPin`, `OutputPin::Error` is `GpioError` as shown:
```rs
impl<'d, T: Pin, MODE> embedded_hal_0_2::digital::v2::OutputPin for PinDriver<'d, T, MODE>
where
    MODE: OutputMode,
{
    type Error = GpioError;

    fn set_high(&mut self) -> Result<(), Self::Error> {
        self.set_level(Level::High).map_err(to_gpio_err)
    }

    fn set_low(&mut self) -> Result<(), Self::Error> {
        self.set_level(Level::Low).map_err(to_gpio_err)
    }
}
```

Fixing the error type is pretty easy but it will obviously be a breaking change, any thoughts?

Vollbrecht, 2023-06-13:
this was a bug that was introduced when the GpioError was implemented for e-hal alpha 10. The e-hal 0.2 versions will still use the EspError. The fix is up in master https://github.com/esp-rs/esp-idf-hal/commit/ec7dafa1b9d7dbb0eae126283d501a9af7d7ea56

leungyauming, 2023-06-14:
Oic, thank you!

closed 2023-06-14 by leungyauming