esp-rs/esp-idf #268: Cannot figure out how to use GPIO interrupts in std build on esp32

nmarks99, 2023-06-23:
I've spent a while trying to figure out the easiest way to use interrupts in my esp-idf-hal project however I have not found a good solution and there are no examples on how to do this. I've seen #144 and #177 but neither of these arrive at a great solution. I'm about to do what was done in #177 and use the lower level esp_idf_sys calls to implement gpio interrupts however I would like to know is a better solution exists or what the idiomatic way to go about this would be.

The only example I have been able to find on using a gpio interrupt in a std environment is the [button-interrupt example](https://github.com/esp-rs/std-training/tree/main/advanced/button-interrupt) in the std-training repo, which builds when using a RISCV target, however for my esp32 (non-riscv) chip as target, I get the following linker error I have not been able to resolve 
`dangerous relocation: l32r: literal target out of range (try using text-section-literals)`

Finally, one more semi related question I have is, is it possible to use [esp32-hal](https://github.com/esp-rs/esp-hal/tree/main/esp32-hal) as a dependency in a std build (with esp-idf-hal)? I was having a look at the [gpio interrupt](https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/gpio_interrupt.rs) example for a no_std build and it seems very straightforward, however I tried (perhaps naively) to simply include the esp32-hal crate in my Cargo.toml and I get the following linker error:
`note: some 'extern' functions couldn't be found; some native libraries may need to be installed or have their path specified`
Any idea on how to fix this or if this is even a reasonable approach?

Any help/suggestions would be greatly appreciated.

ivmarkov, 2023-06-23:
In backwards order:
* `esp32-hal` is unsupported; you probably meant the newer baremetal `esp-hal` crates. In any case, I would recommend against mixing these. You either go with ESP IDF, or without
* There are no good examples (yet), but GPIO interrupts are complete supported, both [as callbacks](https://github.com/esp-rs/esp-idf-hal/blob/master/src/gpio.rs#L1066) as well as with [async](https://github.com/esp-rs/esp-idf-hal/blob/master/src/gpio.rs#L1508). You might be the one contributing the first GPIO interrupts' example
* Join the [matrix chat](https://app.element.io/#/room/#esp-rs:matrix.org)

closed 2023-06-23 by ivmarkov

nmarks99, 2023-06-23:
Thanks, I'll check those out. Is there an example anywhere on how to use the callbacks?

ivmarkov, 2023-06-23:
Not really sure what exactly you anticipate from an example (calling `subscribe` should be pretty trivial), but here's [one](https://github.com/ivmarkov/ruwm/blob/f322dc49392ed98ccda0145d2a932881b5eea133/ruwm-esp32/src/services.rs#L381).

ivmarkov, 2023-06-23:
What I would suggest (again) is to join the Matrix channel, and ask your questions there. GitHub issue tracker is not the most ergonomic medium for this type of interaction.

nmarks99, 2023-06-23:
will do thank you!