esp-rs/esp-idf #270: cargo build fails on newer versions

oyvindnetland, 2023-06-26:
I wanted to play around with this project as im learning rust and like to use ESP32. But i had a few snags when trying to build this, so hope my investigation can be helpful to fix these issues, or possibly figure out what the problem with my setup is.

First of all i didnt get the `cargo espflash` command in the readme to work (didnt recognice --target). So i ended up using 
```
ESP_IDF_VERSION=release/v4.4 cargo build --target xtensa-esp32-espidf --example ledc-simple
```
which at least build the same. I dont know if this is because commands have changed or weirdness with my setup. If its the first, i could make a PR for this.

Secondly, I get this with newer versions:
```
Compiling esp-idf-hal v0.40.1 (/home/oyvindne/esp-rs/esp-idf-hal)
LLVM ERROR: Unexpected operand value!
error: could not compile `esp-idf-hal` (lib)
warning: build failed, waiting for other jobs to finish...
```

I did a bisect, and found that this started with commit dafacbeae9a9ede7c4f72b104ad687b23c750b7

This is with ubuntu 23.04, but i experimented with some other linux versions with docker and seems to be the same

Vollbrecht, 2023-06-26:
Your second error is related to a problem in the compiler-chain - we also observed that in our CI. A fix is already in the making by the compiler folks but will land only in the next release there, so you should downgrade it till than. 
info: https://github.com/esp-rs/esp-idf-hal/commit/73327076186d5603007c6ba6b4382b36c0a3e3c1
you can downgrade by using `espup install --toolchain-version 1.69.0`

For the first problem, what version of espflash are you using? There are some syntax changes between 1.7 and the new 2.0 release versions.

Vollbrecht, 2023-06-26:
espflash 2.0 now needs the `flash` keyword following after `espflash` . For more checkout `espflash --help`

dacut, 2023-07-26:
This should be fixed, I believe, with the new 1.71 toolchain.

MabezDev, 2023-07-27:
Correct, this is fixed in the 1.70.1 release. Closing.

closed 2023-07-27 by MabezDev