esp-rs/esp-idf #275: Would it be useful to have a way to get gpio objects from ints

oyvindnetland, 2023-07-01:
When working on https://github.com/esp-rs/esp-idf-hal/pull/274, I found that it would be nice to have a way to get a gpio pin object from an int (and possible a string). I did not find a way to do this, something i missed, intentional?

```rs
let peripherals = Peripherals::take().unwrap();
let gpio = peripherals.pins.gpio_from_int(1);
```

This would be useful since you could define gpio pins of your project in env or as const in the top of your file. I didnt find an easy way to do stuff like that with the current solution, again, assuming i havent missed something.

I think i have a way of setting it up, just want to check that it makes sense.

MabezDev, 2023-07-03:
You can use [`AnyIOPin`](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/gpio/struct.AnyIOPin.html) to do it dynamically at run time, or if you know the type at compile time you can just call `steal`.

closed 2023-07-03 by MabezDev

Vollbrecht, 2023-07-03:
One more thing i like to add, to get a clean version control and have your hardware-setup codified, you can still use flags for "different boards" you use in one codebase.
One example how you can structure your peripheral base is to do something like the following:
https://github.com/ivmarkov/ruwm/blob/master/ruwm-esp32/src/peripherals.rs 
This has the benefit to allow for different versions marked by a flag, but still retain the complete compiler help with types, don't add real world overhead like a runtime int check would do.

oyvindnetland, 2023-07-05:
hmm, i think i would have imagined something that returned an option to an AnyIOPin or something similar. So it would check if the request was valid for the given chip, and not be an unsafe operation. But i guess the current solution is fine. its possible that what im thinking of would offer too much flexibility to "try" to use different pins, which one could argue should be known and static.