esp-rs/esp-idf #282: esp-idf-hal fails to compile with --cfg esp32c2

dacut, 2023-07-26:
Looks like we have some breakage with the cost-reduced -C2 variant. Not surprising, since it's an oddball.

```
error[E0428]: the name `chip` is defined multiple times
    --> src/gpio.rs:2441:1
     |
2272 | mod chip {
     | -------- previous definition of the module `chip` here
...
2441 | mod chip {
     | ^^^^^^^^ `chip` redefined here
     |
     = note: `chip` must be defined only once in the type namespace of this module

error[E0432]: unresolved import `crate::adc::ADC2`
    --> src/gpio.rs:2291:28
     |
2291 |     use crate::adc::{ADC1, ADC2};
     |                            ^^^^
     |                            |
     |                            no `ADC2` in `adc`
     |                            help: a similar name exists in the module: `ADC1`

error[E0433]: failed to resolve: could not find `ADC2` in `adc`
   --> src/peripherals.rs:213:24
    |
213 |             adc2: adc::ADC2::new(),
    |                        ^^^^
    |                        |
    |                        could not find `ADC2` in `adc`
    |                        help: a struct with a similar name exists: `ADC1`

error[E0412]: cannot find type `ADC2` in module `adc`
   --> src/peripherals.rs:70:20
    |
70  |     pub adc2: adc::ADC2,
    |                    ^^^^ help: a struct with a similar name exists: `ADC1`
    |
   ::: src/lib.rs:221:9
    |
221 |         pub struct $type(::core::marker::PhantomData<*const ()>);
    |         --------------------------------------------------------- similarly named struct `ADC1` defined here

error[E0308]: mismatched types
    --> src/gpio.rs:2473:51
     |
2473 |       pub(crate) static PIN_INTER: [AtomicU8; 20] = [
     |  ___________________________________________________^
2474 | |         super::pin_inter_new(),
2475 | |         super::pin_inter_new(),
2476 | |         super::pin_inter_new(),
...    |
2495 | |         super::pin_inter_new(),
2496 | |     ];
     | |_____^ expected an array with a fixed size of 20 elements, found one with 22 elements
     |
     = note: expected array `[AtomicU8; 20]`
                found array `[AtomicU8; 22]`

error[E0063]: missing fields `carrier_duty_percent`, `carrier_freq_hz`, `carrier_level` and 1 other field in initializer of `esp_idf_sys::rmt_rx_config_t`
   --> src/rmt.rs:997:28
    |
997 |                 rx_config: rmt_rx_config_t {
    |                            ^^^^^^^^^^^^^^^ missing `carrier_duty_percent`, `carrier_freq_hz`, `carrier_level` and 1 other field
```

Assign this to me; I'll take a look at it, see if we can/should add special rules in the CI for this (right now, all RISC-V-based systems are handled the same).

ivmarkov, 2023-09-25:
@dacut Is this still happening?

dacut, 2023-09-25:
@ivmarkov I was just about to check on this. I believe so; it requires some conditional compilation checks to be added.

dacut, 2023-09-25:
Current issues:

```
error[E0425]: cannot find value `SOC_ADC_DIGI_DATA_BYTES_PER_CONV` in this scope
   --> src/adc.rs:815:45
    |
815 |                         max_store_buf_size: SOC_ADC_DIGI_DATA_BYTES_PER_CONV
    |                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `SOC_ADC_DIGI_DATA_BYTES_PER_CONV` in this scope
   --> src/adc.rs:818:42
    |
818 |                         conv_frame_size: SOC_ADC_DIGI_DATA_BYTES_PER_CONV
    |                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `soc_module_clk_t_SOC_MOD_CLK_PLL_F160M` in crate `esp_idf_sys`
     --> src/i2s.rs:111:34
      |
111   |                     esp_idf_sys::soc_module_clk_t_SOC_MOD_CLK_PLL_F160M,
      |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a constant with a similar name exists: `soc_module_clk_t_SOC_MOD_CLK_PLL_F60M`
      |
     ::: /home/dacut/projects/esp-idf-hal/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c3dd5339a5136806/out/bindings.rs:14726:1
      |
14726 | pub const soc_module_clk_t_SOC_MOD_CLK_PLL_F60M: soc_module_clk_t = 6;
      | ----------------------------------------------------------------- similarly named constant `soc_module_clk_t_SOC_MOD_CLK_PLL_F60M` defined here
```

dacut, 2023-09-25:
Fixed in https://github.com/esp-rs/esp-idf-hal/pull/311.

closed 2023-09-26 by ivmarkov