esp-rs/esp-idf #285: Possible use after free in TxRmtDriver

schurwanzn-stabl, 2023-08-04:
When `TxRmtDriver::write_items` is called with `block = false`
https://github.com/esp-rs/esp-idf-hal/blob/29806afcd3fc87759825d4da6895e473ca0c4e68/src/rmt.rs#L607-L613
the called function [`rmt_write_items`](https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32/api-reference/peripherals/rmt.html#_CPPv415rmt_write_items13rmt_channel_tPK12rmt_item32_tib) will keep reading from the passed pointer until the transmission is completed. The borrow on the `S: Signal` though "expires" once `write_items` returns.
For example when calling the `TxRmtDriver::start` function
https://github.com/esp-rs/esp-idf-hal/blob/29806afcd3fc87759825d4da6895e473ca0c4e68/src/rmt.rs#L589-L597
the ESP-IDF may access dropped memory.

---

Aside from that the documentation of `rm_write_items` states to "please [?] do not use the memory allocated from psram when calling rmt_write_items". This restriction is not enforced either.
Should I create a separate issue for this?

ivmarkov, 2023-08-10:
Can you actually contribute a Pr that fixes this? I admit I did not review carefully the RMT driver at the time, and it was a contribution.

schurwanzn-stabl, 2023-09-08:
Fixing the code seems a bit more involved and we currently do not use the non-blocking function.
Thus I can't contribute a fix for now, sorry.