esp-rs/esp-idf #286: Measure of stack use at runtime

LarryMerino, 2023-08-08:
I have the following code: 
```rust
fn main() -> Result<(), InitError> {
    ...
    let _mid_prio_execution = schedule::<8, _>(8000, move || {
            let executor = EspExecutor::new();
            let mut tasks = heapless::Vec::new();

            executor
                .spawn_local_collect(ble_receive_task(z), &mut tasks)
                .unwrap();

            executor
                .spawn_local_collect(ble_send_task(y), &mut tasks)
                .unwrap();

            Ok((executor, tasks))
    });
    ...
}

pub fn schedule<'a, const C: usize, M>(
    stack_size: usize,
    spawner: impl FnOnce() -> Result<(Executor<'a, C, M, Local>, heapless::Vec<Task<()>, C>), SpawnError>
        + Send
        + 'static,
) -> std::thread::JoinHandle<()>
where
    M: Monitor + Wait + Default,
{
    std::thread::Builder::new()
        .stack_size(stack_size)
        .spawn(move || {
            let (executor, tasks) = spawner().unwrap();

            executor.run_tasks(|| true, tasks);
        })
        .unwrap()
}
```

The think is that I need a way to know how much stack is using each task in runtime.
I mean, I need to know how much stack is using in bytes
```rust
ble_receive_task(z)
```
and
```rust
ble_send_task(y)
```
in runtime

🤔 Is there any way to do this?

Sorry if this is a trivial question, but I spent most of the day trying to figure out some way to achieve this, but no luck so far.

ivmarkov, 2023-08-10:
Please use the Matrix chat for such generic questions, rather than opening issue requests :)

With that said, what had worked for me is a simple `core::mem::size_of_val(ble_receive_task(y))`.

There's also something called cargo-call-stack - you can try to google it.

closed 2023-08-10 by ivmarkov

LarryMerino, 2023-08-11:
Thank you
I'll try with this. 
I usually use
```rust
unsafe {
        println!(
            "TOTAL heap free size: {}",
            esp_idf_sys::heap_caps_get_free_size(MALLOC_CAP_8BIT)
        );
    }
```
And normally it works fine for my purpose, but when I use it inside the async function handled by the Executor it gives me wrong data and I need this information to debug the memory usage at runtime because I am running out of memory.
Anyway, thanks again. I'll try this approach and I'm sorry to put these question here.