esp-rs/esp-idf #289: Linking failure for some pthread stuff

stevefan1999-personal, 2023-08-15:
Minimal reproduction case: 

1. Clone https://github.com/jasta/esp32-tokio-demo
2. Add `sync` feature to tokio
3. `multiple definition of` pthread stuff

I'm not sure why `<PROJECT_DIR>/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imc_zicsr_zifencei/ilp32/no-rtti/libnosys.a` is linking with `esp-idf/pthread/libpthread.a`...Plus some pthread functions are implemented in ESP-IDF but they also showed up as not implemented and will always fail.

```
         STDERR OUTPUT:
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libstd-10768515486ad699.rlib(std-10768515486ad699.std.cc51b25a40f24f06-cgu.14.rcgu.o): in function `std::sys::unix::thread::Thread::new':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/thread.rs:91: warning: pthread_attr_destroy is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread.c.obj): in function `pthread_attr_destroy':
          /home/steve/rust/esp32c3-hw/.embuild/espressif/esp-idf/v5.1/components/pthread/pthread.c:769: warning: pthread_attr_init is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libstd-10768515486ad699.rlib(std-10768515486ad699.std.cc51b25a40f24f06-cgu.14.rcgu.o): in function `std::sys::unix::thread::Thread::new':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/thread.rs:62: warning: pthread_attr_setstacksize is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libesp_idf_svc-03063abf942faa1f.rlib(esp_idf_svc-03063abf942faa1f.esp_idf_svc.22e48e3b855e1320-cgu.12.rcgu.o): in function `esp_idf_svc::private::mutex::RawCondvar::notify_all':
          /home/steve/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.2/src/private/mutex.rs:107: warning: pthread_cond_broadcast is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libesp_idf_svc-03063abf942faa1f.rlib(esp_idf_svc-03063abf942faa1f.esp_idf_svc.22e48e3b855e1320-cgu.12.rcgu.o): in function `<esp_idf_svc::private::mutex::RawCondvar as core::ops::drop::Drop>::drop':
          /home/steve/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.2/src/private/mutex.rs:123: warning: pthread_cond_destroy is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_cond_var.c.obj): in function `s_check_and_init_if_static':
          /home/steve/rust/esp32c3-hw/.embuild/espressif/esp-idf/v5.1/components/pthread/pthread_cond_var.c:45: warning: pthread_cond_init is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libesp_idf_svc-03063abf942faa1f.rlib(esp_idf_svc-03063abf942faa1f.esp_idf_svc.22e48e3b855e1320-cgu.12.rcgu.o): in function `esp_idf_svc::private::mutex::RawCondvar::notify_one':
          /home/steve/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.2/src/private/mutex.rs:102: warning: pthread_cond_signal is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_cond_var.c.obj): in function `pthread_cond_wait':
          /home/steve/rust/esp32c3-hw/.embuild/espressif/esp-idf/v5.1/components/pthread/pthread_cond_var.c:94: warning: pthread_cond_timedwait is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libtokio-8b2190f659d78dee.rlib(tokio-8b2190f659d78dee.tokio.766869848cda977c-cgu.06.rcgu.o): in function `std::sys::unix::locks::pthread_condvar::Condvar::wait':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/locks/pthread_condvar.rs:117: warning: pthread_cond_wait is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libstd-10768515486ad699.rlib(std-10768515486ad699.std.cc51b25a40f24f06-cgu.14.rcgu.o): in function `std::sys::unix::thread::Thread::new':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/thread.rs:87: warning: pthread_create is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libstd-10768515486ad699.rlib(std-10768515486ad699.std.cc51b25a40f24f06-cgu.14.rcgu.o): in function `<std::sys::unix::thread::Thread as core::ops::drop::Drop>::drop':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/thread.rs:282: warning: pthread_detach is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libtracing_core-d037a7f402bef1a5.rlib(tracing_core-d037a7f402bef1a5.tracing_core.7ed93c24001b13c2-cgu.4.rcgu.o): in function `std::sys::unix::thread_local_key::get':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/thread_local_key.rs:22: warning: pthread_getspecific is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/libstd-10768515486ad699.rlib(std-10768515486ad699.std.cc51b25a40f24f06-cgu.14.rcgu.o): in function `std::sys::unix::thread::Thread::join':
          /home/steve/snap/rustup/common/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/unix/thread.rs:263: warning: pthread_join is not implemented and will always fail
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/deb [... cut ...] esp32c3-hw/.embuild/espressif/esp-idf/v5.1/components/pthread/pthread_local_storage.c:196: first defined here
          /home/steve/rust/esp32c3-hw/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: warning: /home/steve/rust/esp32c3-hw/target/riscv32imc-esp-espidf/debug/deps/esp32c3_hw-a2d40ebb7011c1f9 has a LOAD segment with RWX permissions
          collect2: error: ld returned 1 exit status
```

stevefan1999-personal, 2023-08-15:
Hmm. I changed `esp-idf-hal` to the latest, set `ESP_IDF_VERSION` to `master` and it worked. Closing this as a false positive, but still a weird issue nonetheless.

closed 2023-08-18 by ivmarkov

LinusU, 2023-12-27:
I'm running into the same issue as well, but I'm trying to use `smol` (or specifically, `async-io` & `async-executor`) and not Tokio.

Changing to `ESP_IDF_VERSION` to `master` instead gives the following error:

```text
error[E0425]: cannot find value `g_wifi_feature_caps` in this scope
   --> /Users/linus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.47.3/src/wifi.rs:498:36
    |
498 |             feature_caps: unsafe { g_wifi_feature_caps },
    |                                    ^^^^^^^^^^^^^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0425`.
error: could not compile `esp-idf-svc` (lib) due to previous error
```

-----

Steps to reproduce:

1. Generate a new project `cargo generate esp-rs/esp-idf-template cargo`
2. Add async-io `cargo add async-executor async-io`
3. Modify `main.rs` to add simple usage of async
    ```rs
    use std::time::Duration;

    use async_io::Timer;
    use async_executor::LocalExecutor;

    fn main() {
        // It is necessary to call this function once. Otherwise some patches to the runtime
        // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
        esp_idf_svc::sys::link_patches();

        // Bind the log crate to the ESP Logging facilities
        esp_idf_svc::log::EspLogger::initialize_default();

        log::info!("Hello, World!");

        let executor = LocalExecutor::new();

        async_io::block_on(executor.run(async {
            loop {
                Timer::after(Duration::from_secs(5)).await;
                log::info!("Hello, loop!");
            }
        }))
    }
    ```
4. Try to build `cargo build --release`
5. Observer "multiple definition of" pthread stuff

Vollbrecht, 2023-12-27:
please don't use esp-idf master branch as it is absolutely only for active development, its api is unstable and we don't make any efforts here in esp-idf-sys to have it reliable build against it.

LinusU, 2023-12-27:
(ping @ivmarkov, would you be able to reopen this one please? 🙏)

LinusU, 2023-12-27:
@Vollbrecht to clarify, the error and steps to reproduce happens on 5.1.1, I only included the part about `master` because it was suggested as a workaround

Vollbrecht, 2023-12-27:
the latest current stable release is v5.1.2 can you test against it?

ivmarkov, 2023-12-27:
> (ping @ivmarkov, would you be able to reopen this one please? 🙏)

I'm not reopening until it is confirmed that the issue is in `esp-idf-sys`. Your steps to reproduce are imprecise, in that:
* you don't mention what versions of `async-io` and `async-executor` you are trying to use.
* i'm unsure of your ESP IDF version
* Unclear what MCU/chip

Rather than doing back-and-forth here, please publish a mini project that reproduces the error somewhere.

ivmarkov, 2023-12-27:
BTW the CI of [this](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/Cargo.toml) does pass (MCU=esp32c3), and it **does** use what you are trying to use. ESP IDF 4.4

LinusU, 2023-12-27:
Good points, I've uploaded a repo now: https://github.com/LinusU/esp32-pthread-issue

> * you don't mention what versions of `async-io` and `async-executor` you are trying to use.

`cargo add` adds the latest version, which currently is `async-executor = "1.8.0"` and `async-io = "2.2.2"`

> * i'm unsure of your ESP IDF version

I used the one from the `esp-rs/esp-idf-template` template, which is currently `ESP_IDF_VERSION = "v5.1.1"`.

> * Unclear what MCU/chip

I'm building for the `esp32c3`.

-----

I'm also building on macOS 14.2.1, using `rustc 1.76.0-nightly (90e321d82 2023-12-02)`. Is there any more information/investigation I could provide?

ivmarkov, 2023-12-27:
No, that's good enough. I'll try to build tmr.

ivmarkov, 2023-12-28:
Remove `"-C", "default-linker-libraries"` from [here](https://github.com/LinusU/esp32-pthread-issue/blob/main/.cargo/config.toml#L8). 

It used to be necessary in the past (helped the linker find some intrinsics), with earlier ESP IDF SDKs, but with newer ESP IDF toolchains seems to be creating problems rather than solving ones.

Not really sure why that happens, but my guess would be that the newer cross toolchains of newer ESP IDFs contain a fake/alternative implementation of `libpthread` (perhaps coming from `newlib` itself?) which should not be there, and which collides with ESP-IDF's custom `libpthread` impl when the linker is told to link against the default libraries of the cross toolchain sysroot.

ivmarkov, 2023-12-28:
@LinusU I've opened (and pinned) an issue [here](https://github.com/esp-rs/esp-idf-template/issues/178).

LinusU, 2023-12-28:
Thank you 🙏 🙏 removing `-C default-linker-libraries` worked around the issue!