esp-rs/esp-idf #294: Need to change PWM frequency of LedcDriver

Angelo13C, 2023-08-28:
Hi!

I need to change the PWM frequency of [LedcDriver](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/ledc/struct.LedcDriver.html) after it has been created because the TMC2209 motor driver needs a clock signal on the STEP pin that moves the motor one step for each rising edge.

The problem now is that after creating a [LedcDriver](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/ledc/struct.LedcDriver.html) there's no way to change its frequency so there's no way to change the speed of the motor.

I thought about adding this:
```rust
/// Set the frequency of the timer.
pub fn set_frequency(&mut self, frequency: Hertz) -> Result<(), EspError> {
    esp!(unsafe {
        ledc_set_freq(self.speed_mode.into(), self.timer.into(), frequency.into())
    })?;
    Ok(())
}
```
in [LedcDriver](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/ledc/struct.LedcDriver.html) but I don't know if it's safe

Vollbrecht, 2023-08-28:
The LEDC hardware is somewhat limited if you want to change the frequency at runtime. That is duo to the fact that each of the 4 timers is shared with multiple channels. So if you are using timer 0 and channel 0 and 1, you will change the frequency of both channels 0 and 1 at the same time because they are bound to timer 0. So what they recommend is : 
- live with that limitation and call it anyway
- swap out the underlying timer with one that is setup with another frequency to the currently used channel
- swap out the channel itself that is using another timer

We don't implement the swapping the channels currently  in the wrapper, if i am not mistaken, so you would need to go with method one currently.

Angelo13C, 2023-08-28:
Ok, but that's not a problem for what I'm trying to do. So I can use the function I wrote above without any problem related to safety, right?