esp-rs/esp-idf #313: Cannot build

hanusek, 2023-10-02:
Hello. I have a problem with build.
```
error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
```
On Friday all was works :|

**Full log error:**
```
  Generated /home/mhanusek/work/code/sandbox/esp32-rust/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-60d9434366124fce/out/build/libespidf.bin
  [881/881] cd /home/mhanusek/work/code/sandbox/esp32-rust/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-60d9434366124fce/out/build/esp-idf/esptool_py && /home/mhanusek/work/code/sandbox/esp32-rust/.embuild/espressif/python_env/idf5.1_py3.10_env/bin/python /home/mhanusek/work/code/sandbox/esp32-rust/.embuild/espressif/esp-idf/v5.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/mhanusek/work/code/sandbox/esp32-rust/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-60d9434366124fce/out/build/partition_table/partition-table.bin /home/mhanusek/work/code/sandbox/esp32-rust/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-60d9434366124fce/out/build/libespidf.bin
  libespidf.bin binary size 0x271b0 bytes. Smallest app partition is 0x100000 bytes. 0xd8e50 bytes (85%) free.
  cargo:root=/home/mhanusek/work/code/sandbox/esp32-rust/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-60d9434366124fce/out
  cargo:rerun-if-changed=/home/mhanusek/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.2/src/include/esp-idf/bindings.h
  cargo:rustc-env=EMBUILD_GENERATED_BINDINGS_FILE=/home/mhanusek/work/code/sandbox/esp32-rust/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-60d9434366124fce/out/bindings.rs

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.1") }
  Using esp-idf v5.1.0 at '/home/mhanusek/work/code/sandbox/esp32-rust/.embuild/espressif/esp-idf/v5.1'
  /home/mhanusek/work/code/sandbox/esp32-rust/.embuild/espressif/esp-idf/v5.1/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
    import pkg_resources
  Compiler supported targets: xtensa-esp32-elf

  Built components: xtensa, esp_ringbuf, efuse, driver, esp_pm, mbedtls, esp_app_format, bootloader_support, bootloader, esptool_py, partition_table, esp_partition, app_update, esp_mm, spi_flash, pthread, esp_system, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, cxx, esp_common, esp_timer, app_trace, esp_event, nvs_flash, esp_phy, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, sdmmc, fatfs, idf_test, ieee802154, json, mqtt, openthread, perfmon, spiffs, ulp, usb, wifi_provisioning
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /home/mhanusek/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/ir/context.rs:530:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ivmarkov, 2023-10-02:
`~/export-esp.sh`

closed 2023-10-02 by ivmarkov

hanusek, 2023-10-10:
@ivmarkov not works.
[Screencast_esp32.webm](https://github.com/esp-rs/esp-idf-hal/assets/3663212/d87919f8-b4a0-4b9d-a81a-b0d9325870a3)