esp-rs/esp-idf #319: error: unknown target triple 'xtensa', please use -triple or -arch

MikeMitterer, 2023-10-08:
I followed the instructions on https://github.com/esp-rs/rust-build and made a project with:
```
# STD Project
cargo generate esp-rs/esp-idf-template cargo
```
No mater if I compile with `cargo build` or if I use `cargo espflash save-image --chip esp32 test`
I get the following error message.

Rust is know for it's helpful error messages but in this case I get a bunch of messages but I don't know what they want to tell me...
A bunch of Env-Vars, CMake-Vars bla, bla, but what is right and what's wrong in this incredible complex build environmen????

The last lines of the error message shows that some flags are wrong or missing - OK, maybe, but where should I set the flags?

Please help!

```
cargo espflash save-image --chip esp32 test
   Compiling esp-idf-sys v0.33.2
error: failed to run custom build command for `esp-idf-sys v0.33.2`

Caused by:
  process didn't exit successfully: `/Volumes/DevLocal/DevRust/Learn/esp32-v2/target/debug/build/esp-idf-sys-6aca48eb5088c401/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/python_env/idf4.4_py3.11_env
  PATH=/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/ninja/1.10.2/:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/python_env/idf4.4_py3.11_env/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/tools:$PATH
  Current system platform: macos-arm64
  Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/python_env/idf4.4_py3.11_env
  PATH=/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/ninja/1.10.2/:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/python_env/idf4.4_py3.11_env/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/tools:$PATH
  cargo:rerun-if-changed=/Volumes/DevLocal/DevRust/Learn/esp32-v2/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_PATH="/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6" IDF_TARGET="esp32" IDF_TOOLS_PATH="/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif" PATH="/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/python_env/idf4.4_py3.11_env/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/ninja/1.10.2/:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/tools:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/mikemitterer/.local/bin:/usr/local/sbin:/opt/homebrew/opt/openjdk@17/bin:/Volumes/DevLocal/DevPython/Production/Utils/src/main/bash:/Volumes/DevLocal/DevPython/Production/Utils/src/main/python:/Volumes/DevLocal/DevBash/Production/BashTools/src:/Users/mikemitterer/bin:/Applications/Sublime Text.app/Contents/SharedSupport/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Users/mikemitterer/.cargo/bin:/Users/mikemitterer/Library/Application Support/JetBrains/Toolbox/scripts:/Users/mikemitterer/Library/Android/sdk/platform-tools:/Users/mikemitterer/Library/Android/sdk/emulator:/Users/mikemitterer/Library/Android/sdk/tools/bin:./node_modules/.bin:/Users/mikemitterer/.yarn/bin:/Users/mikemitterer/.config/yarn/global/node_modules/.bin:/Users/mikemitterer/bin/fmpp_0.9.15/bin::/Users/mikemitterer/.cargo/bin:/Users/mikemitterer/Library/Python/3.11/bin" SDKCONFIG_DEFAULTS="/Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out/gen-sdkconfig.defaults;/Volumes/DevLocal/DevRust/Learn/esp32-v2/sdkconfig.defaults" "cmake" "/Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/python_env/idf4.4_py3.11_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/requirements.txt are satisfied.
  -- Project sdkconfig file /Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out/sdkconfig
  Loading defaults file /Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out/gen-sdkconfig.defaults...
  Loading defaults file /Volumes/DevLocal/DevRust/Learn/esp32-v2/sdkconfig.defaults...
  -- App "libespidf" version: 1
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-da73dfd7aaa6214c/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
  -- Adding linker script /Volumes/DevLocal/DevRust/Learn/esp32-v2/.embuild/espressif/esp-idf/v4.4.6/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
[... cut ...]
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /Users/mikemitterer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/ir/context.rs:530:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ivmarkov, 2023-10-09:
Did you forget to run `./export-esp.sh` prior to building?

ivmarkov, 2023-10-09:
And by the way, and because you mention again how helpful Rust is
> Rust is know for it's helpful error messages but in this case I get a bunch of messages but I don't know what they want to tell me...

... well, this panic is coming from **Rust** bindgen. And it is extremely unhelpful. Yeah, it is not us.

I'll think how (actually if possible at all) to capture this panic and print something meaningful. The crux of the problem is that you are running the wrong Clang (your system one), which does not know that "xtensa" arch exists, as the xtensa arch is not upstreamed yet - neither in Rust, nor in Clang (as these share a common backend - LLVM).

The other problem you complained about in that other thread (the env vars thing) is also an upstream problem - in `cmake-rs`. Which is in the `rust-lang` org, if you notice. I could not find a way to tell `cmake-rs` NOT to pick env vars from the parent process (us). 

So where I'm getting at is that the build is complex, but it is not always us. :)

hanusek, 2023-10-10:
@MikeMitterer 
Can you try:
```
cargo install espup
espup install

$HOME/export-esp.sh
cargo build
```

ivmarkov, 2023-10-10:
Only `~/export-esp.sh` is necessary.

@SergioGasquez This is really a total PITA. Remember [this](https://github.com/esp-rs/esp-idf-template/issues/130#issuecomment-1613002712)?

ivmarkov, 2023-10-10:
> Only `~/export-esp.sh` is necessary.
> 
> @SergioGasquez This is really a total PITA. Remember [this](https://github.com/esp-rs/esp-idf-template/issues/130#issuecomment-1613002712)?

I'll probably bring this for discussion on the next meeting (if I'm able to attend, that is). Or at least leave a reminder about it in the Discussion notes.

I mean, I can try to check for the xtensa Clang lib being on the path when building `esp-idf-sys`, from within the `esp-idf-sys` build, yet having the requirement to the user to run this export script is one more variable in an already complex equation.

ivmarkov, 2023-10-10:
Aaaargh.
@hanusek @MikeMitterer Note that the correct command line (at least on Linux) is:

`. ~/export-esp.sh`

Just doing `~/export-esp.sh` (without the "." and space after it) will **not** do it, as the file seems not to be a real shell script (and its not marked as executable anyway) but just a "variables" source script!

@SergioGasquez Was this always the case, or was this changed just recently?

ivmarkov, 2023-10-11:
For `no_std` - yes. The clang lib (which is the culprit) and which is **the one thing** causing headaches for STD builds needs to be around for the foreseeable future. We don't need the GCC linker. This is only for `no_std`.

SergioGasquez, 2023-10-11:
Right, not sure if you know, and also its definitely not a solution but something that can simplify stuff for std ONLY users, is using the `-s/--std` flag of `espup` which skips installing GCC.

ivmarkov, 2023-10-11:
> Right, not sure if you know, and also its definitely not a solution but something that can simplify stuff for std ONLY users, is using the `-s/--std` flag of `espup` which skips installing GCC.

This is not simplifying anything. We don't use the espup GCC, and we don't care about it, in the path or not. What we **need** is ONLY the `LIBCLANG_PATH` variable. Setup correctly and ideally without a source script users will keep forgetting about.

ivmarkov, 2023-10-11:
And by the way, at the very least we should be improving the message `espup` spits out w.r.t. `export-esp.sh`:
* It is an info log (extra noise). Why?
* It is together with a bunch of other info logs on the console, and difficult to spot. And this is the ONE thing users should not be forgetting to do... :(

SergioGasquez, 2023-10-11:
> This is not simplifying anything. We don't use the espup GCC, and we don't care about it, in the path or not. What we need is ONLY the LIBCLANG_PATH variable. Setup correctly and ideally without a source script users will keep forgetting about.

But even if espup GCC is not used, is still downloaded and installed by espup unless you used that flag. I think setting `LIBCLANG_PATH` automatically can be an improvement. What if we set `LIBCLANG_PATH` in all environments (that means setting it on Unix as its already set for Windows) and we leave the `export-esp.sh` to only modify the `PATH` (this would only be required for no_std users on Unix)

> And by the way, at the very least we should be improving the message espup spits out w.r.t. export-esp.sh:
>
> It is an info log (extra noise). Why?
> It is together with a bunch of other info logs on [...]

The [message saying that you need to source the export file is a warn](https://github.com/esp-rs/espup/blob/main/src/env.rs#L100-L110), this is the usual output in Unix:
```
[2023-10-11T08:26:49Z INFO ] 🔧  Uncompressing tar.xz file to '/home/esp/.rustup/tmp/.tmpWECcbn'
[2023-10-11T08:26:49Z INFO ] 🔧  Installing 'rust-src' component for Xtensa Rust toolchain
[2023-10-11T08:27:35Z INFO ] 🔧  Creating export file
[2023-10-11T08:27:35Z WARN ] 💡  Please, set up the environment variables by running: ' /home/esp/export-esp.sh'
[2023-10-11T08:27:35Z WARN ] ⚠️   This step must be done every time you open a new terminal
[2023-10-11T08:27:35Z INFO ] ✅  Installation successfully completed!
```
Also, in all the installation guides, we mention that Unix users must source the export file, not sure how this can be improved, any suggestion?

ivmarkov, 2023-10-11:
But it's a bunch of **logs** still. Why? This is not a service or a firmware. Look at `rustup`. It does NOT spit out logs, with long timestamps, icons, level whatever - all that noise. Just the essence. And in this essence, the fact that there is something environment related that the user has to pay attention to is the most important, LAST message, on a separate li e and so on. Readable.

SergioGasquez, 2023-10-11:
Here is the output of doing `curl https://sh.rustup.rs -sSf | bash -s -- -y`
```
info: installing component 'rust-std'
 24.7 MiB /  24.7 MiB (100 %)  17.2 MiB/s in  1s ETA:  0s
info: installing component 'rustc'
 61.6 MiB /  61.6 MiB (100 %)  18.9 MiB/s in  3s ETA:  0s
info: installing component 'rustfmt'
info: default toolchain set to 'stable-x86_64-unknown-linux-gnu'

  stable-x86_64-unknown-linux-gnu installed - rustc 1.73.0 (cc66ad468 2023-10-03)

Rust is installed now. Great!

To get started you may need to restart your current shell.
This would reload your PATH environment variable to include
Cargo's bin directory ($HOME/.cargo/bin).

To configure your current shell, run:
source "$HOME/.cargo/env"
```

I created https://github.com/SergioGasquez/espup/tree/feat/environment with some changes to the logs and here is what the output of this branch looks like:
```
[INFO]: 📥  Downloading file '/home/esp/.rustup/tmp/.tmp6taYwd/rust-src.tar.xz' from 'https://github.com/esp-rs/rust-build/releases/download/v1.73.0.0/rust-src-1.73.0.0.tar.xz'
[INFO]: 🔧  Extracting tar.xz file to '/home/esp/.rustup/tmp/.tmp6taYwd'
[INFO]: 🔧  Installing 'rust-src' component for Xtensa Rust toolchain
[INFO]: 🔧  Creating export file

        💡  To get started, you need to set up some environment variables by running: '. /home/esp/export-esp.sh'
        ⚠️  This step must be done every time you open a new terminal.
            See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
[INFO]: ✅  Installation successfully completed!
```

SergioGasquez, 2023-10-11:
Also, I opened https://github.com/esp-rs/espup/pull/373, which simplifies the amount of environment variables that we need to set up. This would be the resulting `export-esp.sh`:
```
export LIBCLANG_PATH="/home/esp/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib"
export PATH="$HOME/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:$PATH"
export PATH="/$HOME/.rustup/toolchains/esp/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:$PATH"
```
We can continue the discussion on the community meeting, but I am happy to set up those variables in `espup` or at least set the `LIBCLANG_PATH` since its less “sensitive” than modifying the user `PATH`

ivmarkov, 2023-10-11:
> I created [esp-rs/espup#375](https://github.com/esp-rs/espup/pull/375) with some changes to the logs and here is what the output of this branch looks like:
> 
> ```
> [INFO]: 📥  Downloading file '/home/esp/.rustup/tmp/.tmp6taYwd/rust-src.tar.xz' from 'https://github.com/esp-rs/rust-build/releases/download/v1.73.0.0/rust-src-1.73.0.0.tar.xz'
> [INFO]: 🔧  Extracting tar.xz file to '/home/esp/.rustup/tmp/.tmp6taYwd'
> [INFO]: 🔧  Installing 'rust-src' component for Xtensa Rust toolchain
> [INFO]: 🔧  Creating export file
> 
>         💡  To get started, you need to set up some environment variables by running: '. /home/esp/export-esp.sh'
>         ⚠️  This step must be done every time you open a new terminal.
>             See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
> [INFO]: ✅  Installation successfully completed!
> ```

This is a **great improvement**, thank you! Would it be possible to put the "To get started, you need..." text as the **last** step / log in the sequence? Basically after "Installation successfully completed!". This way - and similarly to `rustup` - it will be the last thing, and the one action item user needs to care about (on a regular basis).

I guess I'm nit-picking (removing the timestamp was already a big improvement), but do we need the icons? They look cool, but are yet another distraction imo. And `[INFO]` can be lowercased so it is less into the eyes, like e.g. `info:`.

ivmarkov, 2023-10-11:
> ```
> export LIBCLANG_PATH="/home/esp/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-> ```
> 
> We can continue the discussion on the community meeting, but I am happy to set up those variables in `espup` or at least set the `LIBCLANG_PATH` since its less “sensitive” than modifying the user `PATH`

That would be absolutely great, but I suggest rather than putting the variable in `~/.profile` (or whatever), can't we **source** the `export-esp.sh` script from there? My concern is what happens when the user upgrades to a new version of the toolchain, and the value of this variable changes? You need to grep the var in the `~./profile` file and change it, which is risky. Far better to reference `export-esp.sh` from there. And perhaps rename it to `env` (it is not a real `.sh`, is it?) and move it inside `.espup`.

ivmarkov, 2023-10-11:
@SergioGasquez How about the following alternative proposal that completely bypasses altering the user's environment:
* Create a symlink - `$HOME/.espup/libclang_path` - that points to the **actual** directory where the Espressif xtensa liobclang directory is located (as it can change if you upgrade the toolchain with `espup`)
* `esp-idf-sys` should be altered to look for the presence of `$HOME/.espup/libclang_path` and use that if available, or else fallback to the standard mechanism `bindgen` uses to locate `libclang`. This can optionally only be done when building for xtensa targets

This way we kinda get the best of both worlds: 
* Easy way to find the Clang lib location (if installed via `espup`)
* No pollution of user's env 
* Might even work on Windows? Not sure we can symlink there these days?

Sure, `no_std` users would still have to suffer with the `export-esp.sh` script for a while, but that's only because of GCC, and as you say - temporary.

SergioGasquez, 2023-10-11:
> This is a **great improvement**, thank you! Would it be possible to put the "To get started, you need..." text as the **last** step / log in the sequence? Basically after "Installation successfully [...]

Thanks for the feedback! This is how it currently looks like:
```
[warn]: ⚠️   Previous installation of Xtensa Rust 1.73.0.0 exists in: '/home/esp/.rustup/toolchains/esp'. Reusing this installation
[info]: 🔧  Creating export file
[info]: ✅  Installation successfully completed!

        💡  To get started, you need to set up some environment variables by running: '. /home/esp/export-esp.sh'
        ⚠️   This step must be done every time you open a new terminal.
            See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
```

Will think about removing the emojis and the environment variables next week (I'm on vacations until then). And thanks again for all the alternatives!

MikeMitterer, 2023-10-12:
@ivmarkov F...  + SORRY because of another problem `. ~/export-esp.sh` was not called and caused all the error messages. Thanks for pointing me to the right direction. 

@all here in the discussion - Thanks! It's really cool to see how you improve the logs and the whole environment to make the Rust environment, and it's toolchain even more helpful!!!!

ivmarkov, 2023-10-12:
> 
> Will think about removing the emojis and the environment variables next week (I'm on vacations until then). And thanks again for all the alternatives!

[Seems like symbolic links require admin privileges / developer mode on Windows](https://doc.rust-lang.org/stable/std/os/windows/fs/fn.symlink_dir.html).

Well I guess on Windows the `%HOME%\.espup\esp-clang-lib` sym link can just be a copy of the actual directory - `%HOME%\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-16.0.0-20230516\esp-clang\lib`

no?

ivmarkov, 2023-10-17:
(Will keep this open, but renamed the issue to reflect where we got after discussing.)

SergioGasquez, 2023-10-17:
> Might even work on Windows? Not sure we can symlink there these days?

Windows should have this issue already solved, we don't need to source the `export-esp.sh` file on Windows, we modify the environment variables.

> Sure, no_std users would still have to suffer with the export-esp.sh script for a while, but that's only because of GCC, and as you say - temporary.

Even when we use LLD as linker, we would still need to set `LIBCLANG_PATH`, so we would still require the sourcing.

Here is how `rustup` modifies the `PATH`: https://github.com/rust-lang/rustup/blob/master/src/cli/self_update/unix.rs#L83

ivmarkov, 2023-10-17:
> > Might even work on Windows? Not sure we can symlink there these days?
> 
> Windows should have this issue already solved, we don't need to source the `export-esp.sh` file on Windows, we modify the environment variables.
> 

OK, fine. I don't like it because we then do different things for different OS-es, but even if we have the symlink for non-Windows, that would be a step forward.

> > Sure, no_std users would still have to suffer with the export-esp.sh script for a while, but that's only because of GCC, and as you say - temporary.
> Even when we use LLD as linker, we would still need `LIBCLANG_PATH`, so we would still require the sourcing.
> 
> Here is how `rustup` modifies the `PATH`: https://github.com/rust-lang/rustup/blob/master/src/cli/self_update/unix.rs#L83

No. The idea of the symlink is that `esp-idf-sys` would detect it and it will **itself** setup `LIBCLANG_PATH` prior to calling bindgen.

If you don't like the symlink idea - **fine** - __just modify the user's env on ALL operating systems__. Even though this **is** polluting the user env, which was the primary reason I came up with the sourcing link - it is something ONLY `esp-idf-sys` needs to know and use so a symlink is less obtrusive.

But having this "sourcing" for non-Windows is really, really hurting the initial user experience. I absolutely want to avoid this extra step.

SergioGasquez, 2023-10-18:
Im just working on releasing `espup 0.7.0` with GCC 13 (less environment variables to set) and updated logging, since we are probably releasing a `1.73.0.1` Xtensa Rust version. After that I will start working in removing the export file one way or another, also, if we go for the environment approach, we should add a `no-modify-env` flag

ivmarkov, 2023-10-18:
> Im just working on releasing `espup 0.7.0` with GCC 13 (less environment variables to set) and updated logging, since we are probably releasing a `1.73.0.1` Xtensa Rust version. After that I will [...]

Sounds great. If you would like to minimize any changes / impact, we can have the symlink in addition to / without retiring the export file for now. That would be enough for `esp-idf-sys` to pull the libclang location even if the user forgot to source from the export file.

SergioGasquez, 2023-10-23:
Hi! I've been working on this the last few days, and I've come up with two PRs:
- https://github.com/esp-rs/espup/pull/380
- https://github.com/esp-rs/espup/pull/381

Let me know your thoughs, I definitely prefer setting user environment as it will solve the issue for both approaches, and it basically mimics what `rustup` does. In the end, `espup` just tries to be a `rustup` for `esp-rs`.

ivmarkov, 2023-10-23:
My 2c:

The reason why I suggested the symlink approach in the first place - *for both Windows and ?nix though* - is because - in the shiny new world where `espup` will no longer have to install the GCC toolchain, you'll not need to modify the user's env **EXCEPT** for this pesky `libclang` stuff, which - please note - **is only ever used** by `esp-idf-sys`.

So why bother doing rocket science with detecting user shell and whatnot given that this effort will not pay off anymore in the near future? YAGNI.

With that said - as long as I don't have to do the `. ~/export-esp.sh` dance every now and then - and the user env modifications work reliably - I guess I'm fine with either.

SergioGasquez, 2023-10-27:
> My 2c:
>
> The reason why I suggested the symlink approach in the first place - _for both Windows and ?nix though_ - is because - in the shiny new world where `espup` will no longer have to install the [...]

Thanks for your input! Since it seems like there are some progress on LLD side, we can go with the symlink approach, and if we still see some areas for improvements, we can reevaluate later. 

Just to recap, here is how it would like for the different approaches/OSes:
 - Windows currently modifies the user environment, so the user does not need to source the export file
 - Unix would include a symlink from `$HOME/.espup/esp-clang/` to the directory containing Xtensa clang. 
   - This would only help `esp-idf-x` users, so we would keep the documentation as it is (asking the users to source the export file), but if they forget to do so, on `esp-idf-x` it will work anyway.

ivmarkov, 2023-10-27:
Please hold on for a sec.

I'm just debugging an issue, where the build for `esp-idf-sys` fails on ESP IDF `master` because - for whatever reason - the CMake triggered by `esp-idf-sys` seems to pick up the GCC installed by `espup` (instead of the private, hidden, not-on-the`$PATH`, **newer** GCC installed by `esp-idf-sys` itself).

If I can't workaround this by - say - somehow re-ordering the `$PATH` items passed to CMake, we might already have a big problem with the fact that `espup` unconditionally puts a GCC instance in the user env (with `export-esp.sh` for *nix and directly in the user env for Windows).

ivmarkov, 2023-10-28:
@SergioGasquez TL;DR: False alarm. 

Turns out ESP IDF `master` had renamed the `xtensa` GCC toolchain and now has a single one for all `xtensa` MCU variants (just like the `riscv` toolchain, which was always a single one btw).

Yet - and  a bit weirdly - ESP IDF builds tries to use the **old** GCC toolchain (and fails, as it is not compatible with the ESP IDF `master` branch) if it can't locate the new one.

Now, since `espup` is in fact installing an old GCC toolchain *and putting it in the user `$PATH`*, ESP IDF was picking up that one, and failing the build.

ivmarkov, 2023-10-28:
@SergioGasquez I've implemented the symlink support [here](https://github.com/esp-rs/esp-idf-sys/commit/dae898f70e4e4d66ad98ab32b0d4089710444cd1). Let me know in case you change the symlink name and location in the final `espup` release. For now, I assume it is `~/.espup/esp-clang`.

Note that the symlink approach will also run on Windows, where the symlink might not be available, but then why not? User might want to set it up himself there, instead of polluting its environment env.

SergioGasquez, 2023-10-30:
> Now, since espup is in fact installing an old GCC toolchain and putting it in the user $PATH, ESP IDF was picking up that one, and failing the build.

`espup` is also installing the latest GCC now (since [v0.7.0](https://github.com/esp-rs/espup/releases/tag/v0.7.0))

> @SergioGasquez I've implemented the symlink support [here](https://github.com/esp-rs/esp-idf-sys/commit/dae898f70e4e4d66ad98ab32b0d4089710444cd1). Let me know in case you change the symlink name and location in the final espup release. For now, I assume it is ~/.espup/esp-clang.
> 
> Note that the symlink approach will also run on Windows, where the symlink might not be available, but then why not? User might want to set it up himself there, instead of polluting his environment env.

Great, I will do some testing with https://github.com/esp-rs/espup/pull/380 and using the `master` branch to include your commit

SergioGasquez, 2023-10-30:
Update: Just did some testing, and I was able to build an `std` project without sourcing the export file on Linux

SergioGasquez, 2023-11-07:
I think we can now close this issue[image] has been released including this!

closed 2023-11-07 by SergioGasquez