esp-rs/esp-idf #321: adc_cali_curve_fitting_config_t missing chan field

sampaioletti, 2023-10-10:
https://github.com/esp-rs/esp-idf-hal/blob/092055b967c44fe38e79d2eb90b5b17adf8e15af/src/adc.rs#L574

I think the minor version exclusion is wrong, I have v5.1 selected and it doesn't have the chan field

The file at v5.1

https://github.com/espressif/esp-idf/blob/cbce221e88d52665523093b2b6dd0ebe3f1243f1/components/esp_adc/include/esp_adc/adc_cali_scheme.h#L25

Looks like it made it in at v5.1.1
https://github.com/espressif/esp-idf/blob/e088c3766ba440e72268b458a68f27b6e7d63986/components/esp_adc/include/esp_adc/adc_cali_scheme.h#L25

ivmarkov, 2023-10-13:
Would be easier if you just contribute a PR I guess? :)

sampaioletti, 2023-10-13:
I'm happy too.. just wanted to make sure somebody else looked at it before I did in case I was doing something wrong.

-Sam

closed 2023-10-17 by ivmarkov