esp-rs/esp-idf #322: Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

blue-freedom-technologies, 2023-10-11:
### IDF version.

v5.1-beta1-378-gea5e0ff298

### Espressif SoC revision.

ESP32 -C6 (Revision v1.1)

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c6-devkitc-1

### Power Supply used.

USB

### What is the expected behavior?

I expect to be able to use this [subxt_signer](https://docs.rs/subxt-signer/0.32.1/subxt_signer/struct.SecretUri.html) crate.

### What is the actual behavior?

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

### Steps to reproduce.

I am trying to use this  [subxt_signer](https://docs.rs/subxt-signer/0.32.1/subxt_signer/struct.SecretUri.html) crate.

```
use std::str::FromStr;
use subxt_signer::{SecretUri};

let uri = SecretUri::from_str(DEVICE_ACCOUNT_SEEDS).expect("valid URI");
```

### Debug Logs.

```plain
I (40) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (41) boot: compile time Jun  7 2023 08:02:08
I (42) boot: chip revision: v0.0
I (45) boot.esp32c6: SPI Speed      : 40MHz
I (50) boot.esp32c6: SPI Mode       : DIO
I (55) boot.esp32c6: SPI Flash Size : 8MB
I (60) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 007f0000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=c9bd8h (826328) map
I (436) esp_image: segment 1: paddr=000d9c00 vaddr=40800000 size=06418h ( 25624) load
I (449) esp_image: segment 2: paddr=000e0020 vaddr=420d0020 size=68fd8h (430040) map
I (620) esp_image: segment 3: paddr=00149000 vaddr=40806418 size=02224h (  8740) load
I (626) esp_image: segment 4: paddr=0014b22c vaddr=40808640 size=00fc4h (  4036) load
I (633) boot: Loaded app from partition at offset 0x10000
I (634) boot: Disabling RNG early entropy source...
[... 666 lines cut ...]
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x7f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a - wifi_get_target_power
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - _bss_end
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - _bss_end
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - _bss_end
    at ??:??
SHA-256 comparison failed:
Calculated: 377c26481ad6ab0cce508450f353a55f4bd7eb9159aaf85fde3111071f164514
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - _bss_end
    at ??:??
```

ivmarkov, 2023-10-12:
This could be anything. For example, it could be that the crate you are trying to use uses so much RAM, that the measly 300K that you have on the c6 are not enough.

What could help me a bit if you spend a little more time with the  `subxt_signer` crate. As in fork it, clone it and then put some tracing to figure out where it crashes, and whether this is a heap out of memory, or stack too smal (the main task runs with a 4K of stack only!) or something else.

blue-freedom-technologies, 2023-10-13:
Hi,

It blows up exactly here "let uri = SecretUri::from_str(DEVICE_ACCOUNT_SEEDS).expect("valid URI");"

gdb-multiarch view

[image]

/home/silveira/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/alloc/src/collections/btree/node.rs:1031

In [here](https://github.com/rust-lang/rust/blob/master/library/alloc/src/collections/btree/node.rs#L1031)

Cheers,

ivmarkov, 2023-10-13:
**Of course** it blows up on `let uri = SecretUri::from_str(DEVICE_ACCOUNT_SEEDS).expect("valid URI");`. That's the only line referring to that 3rd party crate you have problems using.

Again, the question is a bit: does it blow because of heap mem or stack mem blowup?
Since it is using `BTreeMap` internally - likely heap mem.

You need to kind of learn this crate a bit and then start modifying it to figure out from where the large heap memory usage is coming. Gdb is unlikely to help. Rather, reading the code, commenting bits and pieces of it and the re-compiling and re-flashing.

To mitigate (at least partially) the hypothesis of a stack blowup, you can spawn that line in a separate thread using Rust's `ThreadBuilder` and then assign to that other thread something like 80K stack or so.
 
You can by the way do the exact same excercise on the host (x86/64) machine in a little example program and then limit there the stack size of the spawned thread to see what is the minimum stack size the line from above needs to execute.

blue-freedom-technologies, 2023-10-15:
Hi,

Thank you for your guidance the issue was fixed.

Cheers,

closed 2023-10-15 by blue-freedom-technologies

Vollbrecht, 2023-10-15:
> Hi,
> 
> Thank you for your guidance the issue was fixed.
> 
> Cheers,

It is always nice if you give something more of a background around an answer like "issue was fixed" for other people if they ever come to a similar problem.

blue-freedom-technologies, 2023-10-15:
> > Hi,
> > Thank you for your guidance the issue was fixed.
> > Cheers,
> 
> It is always nice if you give something more of a background around an answer like "issue was fixed" for other people if they ever come to a similar problem.

We wrap-up the code around a [thread](https://doc.rust-lang.org/std/thread/struct.Builder.html) with 32K.

Cheers,