esp-rs/esp-idf #328: Error compiling esp-idf-hal with templateproject generated by esp-idf-template

yanshay, 2023-10-23:
Doing my first steps into ESP32 programming with Rust, and the most basic project doesn't compile, generating the error below.
I found a similar issue reported already some time ago ( #281 ) but it seemed to have been resolved since.
Is it a new bug (hard to believe since it is the most basic flow) or my error somewhere?
Also to add, the esp-template works just fine on my system. 
```
   Compiling esp-idf-hal v0.42.1
error[E0063]: missing field `bclk_div` in initializer of `esp_idf_sys::i2s_pdm_rx_clk_config_t`
   --> /Users/my_user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.1/src/i2s/pdm.rs:127:13
    |
127 |             i2s_pdm_rx_clk_config_t {
    |             ^^^^^^^^^^^^^^^^^^^^^^^ missing `bclk_div`

error[E0425]: cannot find function `xTaskGetIdleTaskHandleForCPU` in this scope
     --> /Users/my_user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.1/src/task.rs:222:9
      |
222   |         xTaskGetIdleTaskHandleForCPU(core as u32)
      |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `xTaskGetIdleTaskHandleForCore`
      |
     ::: /Users/my_user/Trials/esp32/esp-3/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-81b8119a05096114/out/bindings.rs:17095:5
      |
17095 |     pub fn xTaskGetIdleTaskHandleForCore(xCoreID: BaseType_t) -> TaskHandle_t;
      |     ------------------------------------------------------------------------- similarly named function `xTaskGetIdleTaskHandleForCore` defined here

Some errors have detailed explanations: E0063, E0425.
For more information about an error, try `rustc --explain E0063`.
error: could not compile `esp-idf-hal` (lib) due to 2 previous errors

```

Vollbrecht, 2023-10-24:
Please provide with what options you generated the template. Do you also have a manual install of esp-idf on your system?

yanshay, 2023-10-24:
I checked your lead regarding the manual install of esp-idf and found that's the cause. I had the esp-idf PATH active in my shell. I started a new clean shell and now it works fine. I thought I had tried that but seems like I didn't. Thanks!

closed 2023-10-27 by ivmarkov