esp-rs/esp-idf #329: example i2c_master_slave.rs

maayanhaimdulberg, 2023-10-24:
Hi all

The example *i2c_master_slave.rs* uses GPIO22 for I2C
[image]

The thing is, the famous ESP-32-S3-DevKitC-1 and esp32s3 in general, do not have a GPIO22. 
Maybe it is best to change the pins in the example, to suit any kind of esp32?

esp32s3: [https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#i2c] page 460
[image]

esp32: [https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf] page 48
[image]

ivmarkov, 2023-10-25:
PRs welcome!

mdolberg, 2023-10-25:
Challenge accepted. Hope I did it right. First PR in an open source project ever.
https://github.com/esp-rs/esp-idf-hal/pull/330

closed 2023-11-07 by ivmarkov