esp-rs/esp-idf #334: I2S reconfiguration support

bbutkovic, 2023-11-09:
I'm working on a project and I'd like to be able to reconfigure the I2S channels on-the-fly.

The IDF [already has support for that](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv429i2s_channel_reconfig_std_slot17i2s_chan_handle_tPK21i2s_std_slot_config_t) so it would be great if we could also do it here.

I'm willing to work on this, in fact I started already, going to post a WIP PR.

ivmarkov, 2023-11-11:
Can you describe a bit your specific use case, and in particular - why dropping and re-creating the driver would not work for your case?

bbutkovic, 2023-11-13:
I'm writing a sort-of wrapper around the I2S driver that I may or may not publish later as a crate. I hold the driver within my struct wrapper, meaning when I want to drop and re-create the driver I have to also keep ties to the peripherals or pass them into the function that calls for re-configuration. This is all doable but by using the API above it would make it a little bit cleaner and easier to reason about.

skibon02, 2024-07-24:
Hi! I also need reconfiguration possibility in my scenario.

> why dropping and re-creating the driver would not work for your case?

Could you please explain, did you mean using clone_unchecked() for keeping peripherals and reusing them while creating a new instance of I2sDriver with a new configuration?

By making use of `i2s_channel_reconfig_*` functions of esp-idf it would make managing peripherals much easier in code.
Ideally, it'd be great to have `uninit(self)` method which would return participating peripherals as a tuple. But a variety of i2s modes can make it difficult

ivmarkov, 2024-07-24:
> Hi! I also need reconfiguration possibility in my scenario.
> 
> > why dropping and re-creating the driver would not work for your case?
> 
> Could you please explain, did you mean using clone_unchecked() for keeping peripherals and reusing them to create a new instance of I2sDriver with a new configuration?

You can always create the driver with the peripherals passed by `&mut` instead of by moving them in the driver.
This way, if you drop the driver, you can re-use the peripheral in the new instance of the driver.

skibon02, 2024-07-25:
> passed by `&mut` instead of by moving them in the driver

Oh, thanks, i've missed this feature!
But considering reconfig implementation, I can make a PR with attempt to add reconfigure and uninit methods. 

The only thing I want to know is a preferred way to keep I2S mode metadata, are generics ok? (e.g. `I2sDriver<StandardMode, I2sRx>`, `I2sDriver<TdmMode, I2sBiDir>`)

skibon02, 2024-07-25:
I don't want to change mode during reconfiguration, but everything else. And we need to track which mode was selected upon driver creation to call apropriate reconfigure function.

ivmarkov, 2024-07-25:
> I don't want to change mode during reconfiguration, but everything else. And we need to track which mode was selected upon driver creation to call appropriate reconfigure function.

This can be do e in many ways. You can also track it with a simple enum. That is, unless the rexo figure functions s have different signatures for each mode.

ivmarkov, 2026-08-21:
Since a contributor submitted work doing this despite my doubts if we need it (#587 and #597) that would be finally addressed. 
Also my doubts were completely overblown I must say. The change surface was pretty small.

ivmarkov, 2026-08-21:
Fixed now.

closed 2026-08-21 by ivmarkov