esp-rs/esp-idf #338: esp32c3: can peripheral seems misconfigured as of 0.42.0, transmits return ESP_ERR_INVALID_STATE

phansel, 2023-11-16:
Any calls at all to can.transmit(&frame_struct, timeout_u32) will return Err(EspError(259)) aka [0x103 (ESP_ERR_INVALID_STATE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/error-codes.html)) on **esp32c3** as of esp-idf-hal 0.42.0 or later. It's not the second or third call, the very first one will do it. can.receive() also doesn't work.

It does not work as of:
esp-idf-hal 0.42
esp-idf-sys 0.33
embuild 0.31.2

It works as of:
esp-idf-hal 0.41.2
esp-idf-sys 0.33
embuild 0.31.2

This is using roughly the api documented in https://github.com/esp-rs/esp-idf-hal/blob/master/src/can.rs:
```
let peripherals = Peripherals::take().unwrap();
let pins = peripherals.pins;

// filter to accept only CAN ID 881
let filter = can::config::Filter::Standard {filter: 881, mask: 0x7FF };
// filter that accepts all CAN IDs
// let filter = can::config::Filter::standard_allow_all();

let timing = can::config::Timing::B500K;
let config = can::config::Config::new().filter(filter).timing(timing);
let mut can = can::CanDriver::new(peripherals.can, pins.gpio5, pins.gpio4, &config).unwrap();

let tx_frame = can::Frame::new(StandardId::new(0x042).unwrap(), &[0, 1, 2, 3, 4, 5, 6, 7]).unwrap();
can.transmit(&tx_frame, 0_u32);
```

It doesn't appear to matter whether nightly latest or nightly-2023-07-01 is used for esp-idf-hal 0.42, and I can reproduce on aarch64-apple-darwin and -x86_64-unknown-linux-gnu. nightly latest naturally fails to compile with esp-idf-hal 0.41 due to the AtomicU64 ban.

phansel, 2023-11-16:
A workaround to compile against esp-idf-hal 0.41 is to set the toolchain channel to nightly-2023-07-01 or similar.

Vollbrecht, 2023-11-16:
if you checkout the API-difference, before you can use the created can driver, after creating it you need to call `start()?? on it, `start()` and `stop()`are separate methods now. You are in an invalid state because you didn't start it

closed 2023-11-18 by ivmarkov