esp-rs/esp-idf #339: How to use SPI Half-Duplex Mode

Dessera, 2023-11-17:
I'm a beginner of rust embedded development of esp32. I want to drive DS1302 with Half3Wire-Duplex Mode, but code below didn't work.
```rust
use esp_idf_hal::{
    gpio::Gpio2,
    peripherals::Peripherals,
    spi::{
        config::{Mode as SpiMode, Phase, Polarity},
        SpiBusDriver, SpiConfig, SpiDeviceDriver, SpiDriver, SpiDriverConfig,
    },
};

fn main() -> anyhow::Result<()> {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take()?;

    let config = SpiConfig::new()
        .allow_pre_post_delays(true)
        .input_delay_ns(4)
        .duplex(esp_idf_hal::spi::config::Duplex::Half3Wire)
        .cs_active_high();

    let spi_drv = SpiDriver::new(
        peripherals.spi2,
        peripherals.pins.gpio19,
        peripherals.pins.gpio3,
        None::<Gpio2>,
        &SpiDriverConfig::default(),
    )?;

    let mut spi_dev = SpiDeviceDriver::new(spi_drv, Some(peripherals.pins.gpio2), &config)?;

    let mut read_buf = [0u8; 16];
    log::info!("read_buf: {:?}", read_buf);
    spi_dev.transfer(&mut read_buf, &[0x80 | 0x01, 0x00])?;
    log::info!("read_buf: {:?}", read_buf);

    loop {
        esp_idf_hal::delay::FreeRtos::delay_ms(1000);
    }
}
```
The serial output is as follows
```shell
I (363) main_task: Calling app_main()
I (373) eclock_rs: read_buf: [0, 0]
E (373) spi_master: check_trans_valid(792): SPI half duplex mode is not supported when both MOSI and MISO phases are enabled.
I (383) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (393) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (403) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
Error: ESP_ERR_INVALID_ARG
I (413) main_task: Returned from app_main()
```
It seems that I am missing some configurations, but I don't know what I'm missing.

Vollbrecht, 2023-11-17:
Welcome and thanks for reporting.

I think there might be a bug inside the read path of the transfer, when used in half3wire mode , but  have to investigate a bit further.

Dominaezzz, 2023-11-17:
```Rust
        .duplex(esp_idf_hal::spi::config::Duplex::Half3Wire)
```
and
```Rust
    spi_dev.transfer(&mut read_buf, &[0x80 | 0x01, 0x00])?;
```
are incompatible. You can do a transfer with only one wire, you need one wire for reading and one wire for writing.

What you probably what to do there is this.
```Rust
    spi_dev.transaction(&mut [Operation::Read(&mut read_buf), Operation::Write(&[0x80 | 0x01, 0x00]))?;
```

Dessera, 2023-11-18:
> ```rust
>         .duplex(esp_idf_hal::spi::config::Duplex::Half3Wire)
> ```
> 
> and
> 
> ```rust
>     spi_dev.transfer(&mut read_buf, &[0x80 | 0x01, 0x00])?;
> ```
> 
> are incompatible. You can't do a transfer with only one wire, you need one wire for reading and one wire for writing.
> 
> What you probably want to do there is this.
> 
> ```rust
>     spi_dev.transaction(&mut [Operation::Read(&mut read_buf), Operation::Write(&[0x80 | 0x01, 0x00]))?;
> ```
> 
> EDIT:
> 
> Actually, thinking about it a bit more, you're probably wanting to write first. Up to you to decide 😄
> 
> ```rust
>     spi_dev.transaction(&mut [Operation::Write(&[0x80 | 0x01, 0x00]), Operation::Read(&mut read_buf))?;
> ```

I tried to solve the problem with your code, but it returned the same error. When I did not select any Duplex (which should be Full-Duplex at this time), the final read_buf only has 0xF.

Dessera, 2023-11-18:
> ```rust
>         .duplex(esp_idf_hal::spi::config::Duplex::Half3Wire)
> ```
>
> and
>
> ```rust
>     spi_dev.transfer(&mut read_buf, &[0x80 | 0x01, 0x00])?;
> ```
>
> are incompatible. You can't do a transfer with [...]

I tried to solve the problem with your code, but it returned the same error. When I did not select any Duplex (which should be Full-Duplex at this time), the final read_buf only has 0xF.

Dominaezzz, 2023-11-18:
I see. Looks like @Vollbrecht discovered this as well.

From a quick glance it appears we should be setting [this](https://github.com/esp-rs/esp-idf-hal/blob/82a43d637b0758b8a1dfed5cac8ace5cf399683f/src/spi.rs#L1622) to zero in half duplex mode.

[This comment](https://github.com/esp-rs/esp-idf-hal/blob/master/src/spi.rs#L1719) should've been taken care of when half duplex was added in [this PR](https://github.com/esp-rs/esp-idf-hal/pull/143). Though I'm a little surprised no one has encountered this issue yet.

xiaguangbo, 2024-04-01:
I also have the same question,"Half3Wire" how to use?

xiaguangbo, 2024-04-07:
solution: <https://github.com/esp-rs/esp-idf-hal/issues/401>

xiaguangbo, 2024-07-15:
new solution: https://github.com/esp-rs/esp-idf-hal/pull/459

Vollbrecht, 2024-07-18:
Closing as it is now merged! Thanks,

closed 2024-07-18 by Vollbrecht