esp-rs/esp-idf #340: async timer panic

lz1998, 2023-11-17:
```toml
[package]
name = "esp32-test"
version = "0.1.0"
authors = ["lz1998 <875543533@qq.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.47.3", default-features = false }
esp-idf-sys = "0.33"
futures = "0.3"
embassy-futures = "0.1"
esp-idf-hal = {version = "0.42",features = ["nightly"]}
embedded-hal-async = "1.0.0-rc.1"

[build-dependencies]
embuild = "0.31.3"
```

```rust
use esp_idf_hal::timer::TimerConfig;
use esp_idf_svc::eventloop::EspSystemEventLoop;
use esp_idf_svc::hal::peripherals::Peripherals;
use esp_idf_svc::nvs::EspDefaultNvsPartition;
use esp_idf_svc::timer::EspTaskTimerService;
use esp_idf_svc::wifi::{AsyncWifi, AuthMethod, Configuration, EspWifi};
use esp_idf_sys::EspError;

fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello, world!");
    let res = embassy_futures::block_on(init());
    log::info!("{res:?}");
}

async fn init() -> Result<(), EspError> {
    let peripherals = Peripherals::take()?;

    let mut timer = esp_idf_hal::timer::TimerDriver::new(peripherals.timer00, &TimerConfig::new())?;
    for i in 0..100 {
        log::info!("{i}");
        timer.delay(timer.tick_hz()).await?;
    }

    // let sys_loop = EspSystemEventLoop::take()?;
    // let timer_service = EspTaskTimerService::new()?;
    // let nvs = EspDefaultNvsPartition::take()?;
    // let mut wifi = AsyncWifi::wrap(
    //     EspWifi::new(peripherals.modem, sys_loop.clone(), Some(nvs))?,
    //     sys_loop,
    //     timer_service,
    // )?;
    // connect_wifi(&mut wifi).await?;

    Ok(())
}
```

```bash
E (5345) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5345) task_wdt:  - IDLE (CPU 0)
E (5345) task_wdt: Tasks currently running:
E (5345) task_wdt: CPU 0: main
E (5345) task_wdt: Print CPU 0 (current core) registers
Core p8 register dump:
MEPC    : 0x40385a60  RA      : 0x40388f76  SP      : 0x3fc90780  GP      : 0x3fc8ac00  
0x40385a60 - vPortClearInterruptMask
    at /Users/lizheng/RustroverProjects/esp32-test/.embuild/espressif/esp-idf/v5.1.1/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:385
0x40388f76 - __atomic_compare_exchange_4
    at /Users/lizheng/RustroverProjects/esp32-test/.embuild/espressif/esp-idf/v5.1.1/components/newlib/stdatomic.c:313
0x3fc90780 - _btdm_common_start
    at ??:??
0x3fc8ac00 - uxTopUsedPriority
    at ??:??
TP      : 0x3fc7190c  T0      : 0x42033e4a  T1      : 0x42003022  T2      : 0x00000001  
0x3fc7190c - __eh_frame_hdr
    at ??:??
0x42033e4a - core::fmt::Formatter::pad_integral
    at ??:??
0x42003022 - alloc::alloc::dealloc
    at /Users/lizheng/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/src/rust/library/alloc/src/alloc.rs:117
0x00000001 - 
    at ??:??
S0/FP   : 0x00000001  S1      : 0x3fc907a0  A0      : 0x00000001  A1      : 0x3fc907a0  

```

Vollbrecht, 2023-11-17:
can you please specify what esp32 you are using specifically? a normal esp32 running on xtensa?

lz1998, 2023-11-18:
> can you please specify what esp32 you are using specifically? a normal esp32 running on xtensa? Also please clarify what version of the toolchain you are using with `rustc --version`

[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"

lz1998, 2023-11-18:
If I use `esp_idf_hal::task::block_on` instead of `embassy_futures::block_on`, it will not panic. Is it expected?

Vollbrecht, 2023-11-18:
expected? No,
We currently have lots of moving parts with embassy support, there were recently some atomic changes introduces, that also changed inside our crates and also in embassy, this is potentially an outfall here. 

Before recently the most tested way was to use the [edge-executor](https://lib.rs/crates/edge-executor) as a base to run async code, but now we should be in a good position to allow other runners including bigger ones like tokio, with the advent of the possibility to notify waker from ISR context.

@ivmarkov seeing __atomic_compare_exchange_4 in his crash, could this related to the recent changes @Dirbaio made in embassy to help the no_std guys back on track?

ivmarkov, 2023-11-18:
This is not a panic - look at the top of the stack trace and google "esp idf twdt".  And NO you should not be using `embassy_futures::block_on`. Read the comment above that function as to what it is doing, and that would hopefully explain why the task watchdog is triggering.

ivmarkov, 2023-11-18:
> expected? No, We currently have lots of moving parts with embassy support, there were recently some atomic changes introduces, that also changed inside our crates and also in embassy, this is potentially an outfall here.

This behavior is completely normal and expected, as per above.

closed 2023-11-18 by ivmarkov

Vollbrecht, 2023-11-18:
> > expected? No, We currently have lots of moving parts with embassy support, there were recently some atomic changes introduces, that also changed inside our crates and also in embassy, this is potentially an outfall here.
> 
> This behavior is completely normal and expected, as per above.

tanks for pointing it out,  because of all the atomics stuff these days, they magically grab my attention currently making me blindsided :smiling_face_with_tear: