esp-rs/esp-idf #347: need ulp support

zzn26, 2023-12-01:
What I want to achieve is to wake up in deep-sleep state by ulp to detect long key presses.But I can't seem to find any support for ulp.

Vollbrecht, 2023-12-02:
which ulp are you talking about? what esp variant are you using. There are different ulp's for different variants. For example the esp32s3 has the classical ulp/fsm that is programmed in assambly but also has a second variant that can be compiled with a normal riscv toolchain. For the first variant you can find some examples in the  riscv-ulp directory. If you want to see how you can interact with it from your normal application you can have a look inside the std-demo from ivmarkov [here](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L1353).

zzn26, 2023-12-04:
> which ulp are you talking about? what esp variant are you using. There are different ulp's for different variants. For example the esp32s3 has the classical ulp/fsm that is programmed in assambly [...]

@Vollbrecht I'm sorry if I wasn't clear. I'm using esp32s3.I clone the std-demo of ivmarkov you mentioned, but unfortunately cargo check suggests me
```
no method named `load` found for struct `esp_idf_svc::hal::ulp::ULP` in the current scope
    --> src/main.rs:1357:13
```
I'm not sure what's wrong with that

zzn26, 2023-12-05:
I seem to have found the problem, my esp-idf version is 5.1.1 and the ulp compile condition requires a major version number of 4.

zzn26, 2023-12-05:
> you see that he is using a specifc [sdkconfig](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/sdkconfig.defaults.esp32s2) file for the s2 here in the project and he mentiond that this example assume s2. Some of the methods are only availabe if ulp support is enabled via sdkconfigs. You can see the flags that are checked in the source code of [ulp](https://github.com/esp-rs/esp-idf-hal/blob/master/src/ulp.rs#L213). So this needed to be set in the sdkconfigs with the appopriate names simmilar how its done in the demo's sdkconfig

@Vollbrecht I solved this problem by referring to ivmarkov's demo and calling esp-idf-sys directly, thank you very much for your answer.

closed 2023-12-05 by zzn26