esp-rs/esp-idf #350: Not checking for ESP_FAIL in AsyncCanDriver::transmit

leonardodanelutti, 2023-12-04:
For non-blocking implementations of the `embedded-hal` the `ESP_FAIL` error is treated as `WouldBlock`, shouldn't the same be done for the async function implementation?
```rust
pub async fn transmit(&self, frame: &Frame) -> Result<(), EspError> {
    loop {
        let res = self.driver.borrow().transmit(frame, delay::NON_BLOCK);

        match res {
            Ok(()) => return Ok(()),
            Err(e) if e.code() != ESP_ERR_TIMEOUT => return Err(e),
            _ => (),
        }

        WRITE_NOTIFICATION.wait().await;
    }
}
```

ivmarkov, 2023-12-08:
No idea without digging (again) into the ESP IDF C source code of the TWAI driver...

ivmarkov, 2023-12-08:
@leonardodanelutti [The way I read the code](https://github.com/espressif/esp-idf/blob/release/v5.1/components/driver/twai/twai.c#L638) (I might be wrong) is that `ESP_FAIL` would only be returned if you are creating the driver without a TX queue (i.e. `tx_queue_len` == 0)

Can you confirm if you are doing that?

leonardodanelutti, 2023-12-08:
@ivmarkov I actually didn't get that error, I was looking at the async implementation for the CanDriver and it seemed strange that the ESP_FAIL error wasn't also checked. Yes, I think you are right, as it is also written in the documentation, ESP_FAIL is returned only if the TX queue is disabled and another message is transmitting at that moment. When that error occurs, I don't think the function should return, right?

ivmarkov, 2023-12-09:
Yes I'll patch the driver shortly.

ivmarkov, 2023-12-21:
https://github.com/esp-rs/esp-idf-hal/commit/f958b77bf094f1dc42ea12ebacc651ed1a244221

closed 2023-12-21 by ivmarkov