esp-rs/esp-idf #356: ADC_Calibration

morteza1993, 2023-12-30:
Hi, I would like to know where should I active this flag "esp_idf_comp_esp_adc_cal_enabled" for using adc calibration feature within adc.rs file ?
Is there any flag in sdkconfig.defaults ? Can i active this flag "esp_idf_comp_esp_adc_cal_enabled" from rust side itself ?

Vollbrecht, 2023-12-31:
This flags usually gets automatically set if the upstream esp-idf feature is there and enabled. In this case it is dependend on a feature in esp-idf v4 . Since esp-idf v5 they deprecated the api and came up with a new one and i think they also renamed some of the underlying configs.

So if you are using esp-idf v4 you should be able to enable it via sdkconfig. For esp-idf v5 we currently don't have the renamed flags exported i think that are used in the v5 deprecated api. For v5 we have the oneshot and continuous driver apis that were introduced. We automatically instantiate the calibration for oneshot in that api for you if its available [here](https://github.com/esp-rs/esp-idf-hal/blob/master/src/adc.rs#L535). For more information about the underlying general api you can checkout the esp-idf api [description](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc_oneshot.html)

morteza1993, 2023-12-31:
About this part 

> upstream esp-idf feature is there and enabled.
What do you mean  ( Which feature should be activated in upper layer . I tried "esp_idf_comp_esp_adc_cal_enabled" but it was not a feature to be selectable . )

And about this part :

> So if you are using esp-idf v4 you should be able to enable it via sdkconfig

Yes we are working on v4 and I set "esp_idf_comp_esp_adc_cal_enabled=y" in sdkconfig . but nothing changed . 
It's good to point We are using esp32s3 (Xtensa) and it seems in v4 there is not good support for adc calibration for that . 
And at this point we can't upgrade esp-idf version so :
I know there is some functionality in c driver here "https://github.com/espressif/esp-idf/blob/5d8fb28/components/esp_adc/include/esp_adc/adc_cali_scheme.h" 
How can I bring these functionality to my rust project and call it from rust ? should i add something to build system parts ? if yes where and how ?

Vollbrecht, 2023-12-31:
to clarify you cant directly do something like esp_idf_comp_esp_adc_cal_enabled in the sdkconfig, only something that would trigger this to enable it if it can be enabled. For more information about calibrating you need to read the esp-idf v4 adc guide and understand what they do there + how you potential use the fuse system to burn calibration. You will find there, they list the different sdkconfigs that enables them( like ESP_ADC_CAL_VAL_EFUSE_VREF)

Regarding the functionality you found, you cant because what you are linked is part of the newer driver api introduced in esp-idf v5.

Vollbrecht, 2023-12-31:
To checkout what configuration currently are all enabled you can use `cargo rustc -- --print cfg` in your project dir

closed 2024-01-01 by morteza1993