esp-rs/esp-idf #359: esp32 d0wd

lattice0, 2024-01-04:
Hi, what about support for more specific chips like esp32 d0wd? Is this a subset of the ESP32CX ones for example? Will it work?

Vollbrecht, 2024-01-04:
we support every esp32 chip that is supported in the respective upstream ESP-IDF version. esp32 d0wd is just a chip variation of the classical xtensa based esp32, don't confuse them with esp32c* variants that are riscv based.

lattice0, 2024-01-04:
@Vollbrecht thanks I'm starting to get into this world and found out the only cheap LTE supported board (LILYGO® TTGO T-A7670G/E/SA R2 4G) uses esp32 d0wd. So I'd use the ESP32-S2 or S3?

Vollbrecht, 2024-01-04:
to my understanding it should be neither, its just a plain esp32. for more info best place to check is to visit the espressif site directly and check the [soc](https://www.espressif.com/en/products/socs) section for chip level info and the [module](https://www.espressif.com/en/products/modules) section for module level info

ivmarkov, 2024-01-05:
Thanks, @Vollbrecht !

closed 2024-01-05 by ivmarkov