esp-rs/esp-idf #361: Error when compiling fresh project

Flamabalistic, 2024-01-09:
I've just started a new std project, following the instructions in https://esp-rs.github.io/book, and I'm having trouble getting the template to compile.

I've created the template using `cargo generate esp-rs/esp-idf-template cargo` -> esp32s2 -> use default options. I have not made any changes after this.

When running `cargo run`, I receive the following error:
```
error[E0425]: cannot find value `g_wifi_feature_caps` in this scope
   --> C:\Users\torna\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-idf-svc-0.47.3\src\wifi.rs:498:36
    |
498 |             feature_caps: unsafe { g_wifi_feature_caps },
    |                                    ^^^^^^^^^^^^^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0425`.
error: could not compile `esp-idf-svc` (lib) due to previous error
```

Additional notes:
 - I've tried with the `master` branch, yielding the same issue.
 - I've tried with no-std, and it works as expected

Vollbrecht, 2024-01-09:
thanks for reporting, there seams to be some problems with respect to psram. Some updates where made in upstream esp-idf but not backported yet. 

Could you try setting `CONFIG_ESP32_SPIRAM_SUPPORT=n` or `CONFIG_SPIRAM_IGNORE_NOTFOUND=y` in the sdkconfig.default file and rebuild with that options.

We will see in the meantime if we can salvage the situation with maybe some conditional foo added here.

Some meta info -> [here](https://github.com/espressif/esp-idf/commit/87cc516338c034bfdea31983658dde37ed029fac) is long therm fix deployed

Flamabalistic, 2024-01-10:
I tried placing `CONFIG_ESP32_SPIRAM_SUPPORT=n`, and `CONFIG_SPIRAM_IGNORE_NOTFOUND=y` in sdkconfig.defaults, same issue.

I also tried putting just one in and building, then putting just the other in and building, which also didn't work.

I'm relatively new, so just to make sure I'm doing this right, I'm:
-  `cargo clean`
- Add the relevant line to sdkconfig.defaults
- `cargo build`

In case it helps with tracking the problem, it's apparently been around since at least Dec 25 - being reported [here](https://www.reddit.com/r/rust/comments/18q6qlv/cannot_compile_esp32_program/)

SuddenlyHazel, 2024-01-10:
Hitting the same issue with `esp32c3`

As temporary workaround - I found creating a new project from without default settings, and selecting the [4.4](https://github.com/espressif/esp-idf/releases/tag/v4.4.5) version builds successfully.

ivmarkov, 2024-01-10:
Bug-fixing commit - for reference: https://github.com/esp-rs/esp-idf-svc/commit/efa0e75aa0de6d74e61be2ff7e9b5ed11a7c2b17

ivmarkov, 2024-01-10:
By the way folks (@Flamabalistic @SuddenlyHazel) the easiest way to escalate issues like that is the Matrix chat which is linked-in the README of this crate, rather than Reddit.

Flamabalistic, 2024-01-10:
Good to know! (The reddit post isn't mine BTW, I just found it when looking for people with the same issue)

closed 2024-01-18 by ivmarkov