esp-rs/esp-idf #363: Extra component

nOrOuzi1993, 2024-01-16:
Is there an example or something to help us how to add a new component to esp-idf-hal.
I have read "https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#extra-esp-idf-components", but even that is not a complete guide. e.g. I searched for this
`[package.metadata.esp-idf-sys]
extra_components = [
 { component_dirs = [ "dir1", "dir2" ], bindings_header = "bindings.h", bindings_module = "name" }
]`
i added this to my cargo.toml in the root directory, but nothing happens ("I expected to at least get an error message that there is no dir1 dir2 or something similar, but nothing happened after the cargo build).
My real question is that I use the  following
`[dependencies]
esp-idf-svc = { version = "0.47.3", default-features = false }`
And I would like to add the following :
`https://github.com/joltwallet/esp_littlefs`
as an external component to my project.
What are the ways I can achieve this (I prefer to use it as a patch besides using "esp-idf-svc" in my load and not to clone "esp-idf-svc" codes to my project directly and customize this . )

N3xed, 2024-01-16:
See https://github.com/esp-rs/esp-idf-sys/pull/114#issuecomment-1207168854 for reference.

nOrOuzi1993, 2024-01-16:
For remote components,
I've tried this
 <lock>></lock> It should also be noted that with [remote components](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#remote-components-idf-component-registry) you no longer need to provide esp-idf components.

And add this :
`[package.metadata.esp-idf-sys.extra_components.0.remote_component]
# The name of the remote component. Corresponds to a key in the dependencies of
# `idf_component.yml`.
name = "littlefs"
# The version of the remote component. Corresponds to the `version` field of the file
# `idf_component.yml`.
version = "1.13.0"
# A Git URL containing this remote component. Corresponds to the `git` field
# Field of the `idf_component.yml`.
#
# This field is optional.
git = "https://github.com/joltwallet/esp_littlefs"
# A path to the component.
# Corresponds to the `path` field of `idf_component.yml`.
#
# Note: This shouldn't be used for local components, use
# `component_dirs` of extra components instead.
#
# This field is optional.
# path = "path/to/component"
path = "excomponent/small"

# A url to a custom component registry. Corresponds to the `service_url` field
# Field of the `idf_component.yml`.
#
# This field is optional.
# service_url = "https://componentregistry.company.com"
`
in my Cargo.toml in the root directory, but nothing happened after I ran the Cargo build (nothing with the name littlefs came up in the bindings.rs or anywhere else).
Am I missing something?

N3xed, 2024-01-16:
I don't know if you already did this, but you still need a `bindings_header`, so a C header file that includes the relevant littlefs bindings. Otherwise, no bindings to littlefs APIs will be generated. The `bindings.h` in https://github.com/esp-rs/esp-idf-sys/pull/114#issuecomment-1207168854.

esp-idf-sys is probably not rebuilt from the `cargo.toml` changes, so you have to `cargo clean -p esp-idf-sys` to force a rebuild of esp-idf-sys and the generated bindings (this should work to force a rebuild of esp-idf-sys, if not do a whole clean rebuild `cargo clean`).

N3xed, 2024-01-16:
Its also good to compile with `cargo build -vv` (the `-vv` flag) so that the build output of esp-idf-sys is shown, which displays the build components etc.

nOrOuzi1993, 2024-01-16:
Now it is interesting and easy to use .
Thank you .(I tried both ways and got answer from both ) .
And in the second way ("Remote" one) It adds "Components_esp32s3.lock" during building and everything has been built.
```
dependencies:
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.1.1
  joltwallet/littlefs:
    component_hash: f7813d2c2ddde8bc74cd9d044f063d758672f8504ca56132f518ff1438c9a051
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.13.0
manifest_hash: 901bba517a4f6d0ff5dfd02a5aaff14e4eb6faa3517d481644bf6b99992f04fe
target: esp32s3
version: 1.0.0

```

nOrOuzi1993, 2024-01-16:
Thanks to "[N3xed](https://github.com/N3xed)" , Got the result that I wanted.

closed 2024-01-16 by nOrOuzi1993