esp-rs/esp-idf #364: esp_idf_hal::delay::Delay missing some methods for embedded_hal::blocking::delay

mingp, 2024-01-18:
Excuse me that I'm new to both ESP32 and Rust, so I apologize in advance if there is any confusion here.

When writing a small example, I noticed that `esp_idf_hal::delay::Delay`, the recommended delay implementation, does not implement some methods of `embedded_hal::blocking::delay`.

Here is the code for my example, which illustrates what I encountered.

```rust
fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("start");

    let peripherals = esp_idf_hal::peripherals::Peripherals::take().unwrap();
    let i2c_device = peripherals.i2c0;
    let sda_pin = peripherals.pins.gpio20;
    let scl_pin = peripherals.pins.gpio21;

    let i2c_config = esp_idf_hal::i2c::I2cConfig::new().baudrate(10_000u32.into());
    let i2c = esp_idf_hal::i2c::I2cDriver::new(i2c_device, sda_pin, scl_pin, &i2c_config).unwrap();

    struct DelayCompat {
        delay: esp_idf_hal::delay::Delay,
    }
    impl embedded_hal::blocking::delay::DelayUs<u16> for DelayCompat {
        fn delay_us(&mut self, us: u16) {
            self.delay.delay_us(us as u32);
        }
    }
    impl embedded_hal::blocking::delay::DelayMs<u8> for DelayCompat {
        fn delay_ms(&mut self, ms: u8) {
            self.delay.delay_ms(ms as u32);
        }
    }

    let mut delay = DelayCompat {
        delay: esp_idf_hal::delay::Delay::new_default(),
    };
    let mut lcd = hd44780_driver::HD44780::new_i2c(i2c, 0x27u8, &mut delay).unwrap();

    log::info!("initialized");

    lcd.reset(&mut delay).unwrap();
    lcd.clear(&mut delay).unwrap();
    lcd.set_display_mode(
        hd44780_driver::DisplayMode {
            display: hd44780_driver::Display::On,
            cursor_visibility: hd44780_driver::Cursor::Invisible,
            cursor_blink: hd44780_driver::CursorBlink::Off,
        },
        &mut delay,
    )
    .unwrap();
    lcd.write_str("ESP32 + Rust", &mut delay).unwrap();

    log::info!("done");
}
```

This code uses the `hd44780-driver` crate LCD driver, in I2C mode. 

For reference, here are a subset of imports used by `hd44780-driver` and the signature of `HD44780::new_i2c`:

```rust
use embedded_hal::blocking::delay::{DelayMs, DelayUs};

// ...

pub fn new_i2c<D: DelayUs<u16> + DelayMs<u8>>(
        i2c_bus: I2C,
        address: u8,
        delay: &mut D,
    ) -> Result<HD44780<I2CBus<I2C>>>
```

I noticed that `esp-idf-hal`'s default `Delay` implementation, in `esp_idf_hal::delay::Delay`, did not implement the necessary traits. I ended up implementing a shim layer, called `DelayCompat` in the earlier example.

If I look inside `esp-idf-hal`'s `Cargo.toml`, I see that the crate references 2 separate versions of `embedded-hal`, `1.x` by default, and additionally `0.2.7` under a rename.

```toml
embedded-hal = "1"
embedded-hal-0-2 = { package = "embedded-hal", version = "0.2.7", features = ["unproven"] }
```

Notably, I see that a number of variants of `DelayUs` and `DelayMs` are implemented for the `embedded_hal_0_2` `0.2.7` version, but not for the `embedded_hal` `1.x` version.

Should those methods be additionally be implemented?

If the answer is yes, I'd be happy to submit a pull request.

Thank you for your time.

ivmarkov, 2024-01-18:
The above analysis is incorrect. The latest **released** `esp-idf-hal` **also** implements e-hal 1.0.0-rc1 versions of the traits. However, due to Rust lang rules, (a) you need to depend on the e-hal `1.0.0-rc1` crate, with that exact version (b) you need to put a `use` statement for the DelayUs/DelayMs trait in your code.

Note also that RC crate versions and not compatible between each other, i.e. rc1 is not compatible with rc2 and then also not compatible with the final 1.0.0 version which just got released a few days ago.

Note also that you are looking at `esp-idf-hal` MASTER which has nothing to do with the latest released `esp-idf-hal` version.

Finally, please ask the driver author to make a release of his crate which depends on e-hal 1.0.0

The upcoming release of `esp-idf-hal` will only support e-hal 1.0.0, not previous e-hal 1.0.0-rcX versions.

Further questions please direct at the Matrix channel, don't open issues for these please :)

closed 2024-01-18 by ivmarkov

mingp, 2024-01-19:
Thank you for clarifying, and thank you for explaining. Sorry about the earlier confusion. I had believed at the time that this was missing in upstream, which is why I tried to report it as a bug. I appreciate the detailed response.