esp-rs/esp-idf #365: `AsyncCanDriver` is not `Send`

MikiLoz92, 2024-01-18:
I was trying to use the new `AsyncCanDriver` within an Embassy task and found out I am not able to make my code compile, due to the fact that AsyncCanDriver does not implement `Send`:

```
error[E0277]: `*const ()` cannot be shared between threads safely
   --> src/main.rs:186:9
    |
184 |       tasks.push(EX.spawn(
    |                     ----- required by a bound introduced by this call
185 |           Priority::High,
186 | /         can_test_task(
187 | |             peripherals.can,
188 | |             peripherals.pins.gpio7,
189 | |             peripherals.pins.gpio6,
190 | |         )
    | |_________^ `*const ()` cannot be shared between threads safely
    |
    = help: within `AsyncCanDriver<'_, CanDriver<'_>>`, the trait `Sync` is not implemented for `*const ()`
note: required because it appears within the type `PhantomData<*const ()>`
   --> /Users/miki/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/marker.rs:809:12
    |
809 | pub struct PhantomData<T: ?Sized>;
    |            ^^^^^^^^^^^
note: required because it appears within the type `CAN`
   --> /Users/miki/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.5/src/can.rs:934:25
    |
934 | crate::impl_peripheral!(CAN);
    |                         ^^^
note: required because it appears within the type `PeripheralRef<'_, CAN>`
   --> /Users/miki/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.5/src/peripheral.rs:16:12
    |
16  | pub struct PeripheralRef<'a, T> {
    |            ^^^^^^^^^^^^^
note: required because it appears within the type `CanDriver<'_>`
   --> /Users/miki/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.5/src/can.rs:379:12
    |
379 | pub struct CanDriver<'d>(PeripheralRef<'d, CAN>, EnumSet<Alert>);
    |            ^^^^^^^^^
note: required because it appears within the type `AsyncCanDriver<'_, CanDriver<'_>>`
   --> /Users/miki/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.5/src/can.rs:555:12
    |
555 | pub struct AsyncCanDriver<'d, T>
    |            ^^^^^^^^^^^^^^
    = note: required for `&AsyncCanDriver<'_, CanDriver<'_>>` to implement `std::marker::Send`
note: required because it's used within this `async` fn body
   --> /Users/miki/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.42.5/src/can.rs:637:73
    |
637 |       pub async fn transmit(&self, frame: &Frame) -> Result<(), EspError> {
    |  _________________________________________________________________________^
638 | |         loop {
639 | |             let res = self.driver.borrow().transmit(frame, delay::NON_BLOCK);
640 | |
...   |
648 | |         }
649 | |     }
    | |_____^
    = note: required because it captures the following types: `embassy_sync::channel::Receiver<'_, CriticalSectionRawMutex, esp_idf_hal::can::Frame, 8>`, `esp_idf_hal::can::config::Config`, `AsyncCanDriver<'_, CanDriver<'_>>`, `esp_idf_hal::can::Frame`, `impl futures::Future<Output = Result<(), EspError>>`, `embassy_time::Timer`, `impl futures::Future<Output = Result<esp_idf_hal::can::Frame, EspError>>`, `embassy_time::Timer`
note: required because it's used within this `async` fn body
   --> src/peripherals/can/mod.rs:35:25
    |
35  |   ) -> anyhow::Result<()> {
    |  _________________________^
36  | |
37  | |     let commands_recv = CAN_COMMANDS_CHANNEL.receiver();
38  | |
...   |
68  | |
69  | | }
    | |_^
note: required by a bound in `PriorityExecutor::<'a>::spawn`
   --> src/executor.rs:35:43
    |
32  |     pub fn spawn<T: Send + 'a>(
    |            ----- required by a bound in this associated function
...
35  |         future: impl Future<Output = T> + Send + 'a,
    |                                           ^^^^ required by this bound in `PriorityExecutor::<'a>::spawn`
```

However, I see the following code has been implemented for `CanDriver`, within `esp-idf-hal`:
```
unsafe impl<'d> Send for CanDriver<'d> {}
```

Is there a reason that this `impl` is not done for AsyncCanDriver as well?

ivmarkov, 2024-01-18:
No, I have simply forgotten about it. Will fix shortly.

ivmarkov, 2024-01-18:
But that's still strange. I don't know `embassy-executor` in detail, but why is it requiring the Future-s to be `Send`? After all, they ALL run in a single thread, so to say. This is weird?

ivmarkov, 2024-01-18:
Can you paste your code here? Or upload it somewhere? I suspect there is something else fishy going on (not that the async driver should not be `Send`, but with embassy, that should not be a prob).

MikiLoz92, 2024-01-18:
Oh we're not using embassy-executor, but actually using your `edge-executor` crate! We're using the same [PriorityExecutor](https://github.com/ivmarkov/edge-executor/blob/master/examples/priority.rs) defined under the examples, as a matter of fact. And that does require `Send`.

In fact, since we're using just one thread to run all our tasks, I just removed the `Send` requirement for the futures under that executor, and made it use `LocalExecutors`, and now it compiles:

```
pub struct PriorityExecutor<'a> {
    ex: [LocalExecutor<'a>; 3],
}

unsafe impl<'a> Sync for PriorityExecutor<'a> {}

impl<'a> PriorityExecutor<'a> {

    pub const fn new() -> PriorityExecutor<'a> {
        PriorityExecutor {
            ex: [LocalExecutor::new(), LocalExecutor::new(), LocalExecutor::new()],
        }
    }

    pub fn spawn<T: 'a>(
        &self,
        priority: Priority,
        future: impl Future<Output = T> + 'a,
    ) -> Task<T> {
        self.ex[priority as usize].spawn(future)
    }

    pub async fn run(&self) {
        loop {
            for _ in 0..200 {
                let t0 = self.ex[0].tick();
                let t1 = self.ex[1].tick();
                let t2 = self.ex[2].tick();

                t0.or(t1).or(t2).await;
            }

            future::yield_now().await;
        }
    }
}
```

ivmarkov, 2024-01-18:
Yep, exactly! I can't imagine anybody using the parallel executor in embedded. I mean, if you really need to utilize multiple threads, you are probably better off with a separate `LocalExecutor` in each thread, and then if you need the futures in these executors to share stuff, use `Signal` or `Channel` or something from `embassy-sync`. And then only the futures of the `embassy-sync` synchronization primitives need to be `Send` (and you can easily achieve this with using them with a `CriticalSectionRawMutex` or your own `RawMutex` impl based on `std::sync::Mutex`.), but not the rest.

commit 2b4b7206 (2024-01-26): Fix #365
closed 2024-01-26 by ivmarkov