esp-rs/esp-idf #372: Documentation/API regarding default input pin pull up/down state

jobafr, 2024-01-26:
Hi,

I'm currently in the process of porting some Arduino based ESP32 firmwares to Rust. On one device, we noticed that one of the input pins always read as high with the Rust firmware, even in situations where it should have been low, and where the Arduino based firmware reads it as low. After some digging, I found the `PinDriver::set_pull` method. By explicitly calling `pin.set_pull(Pull::Floating)` after creating the `PinDriver` object, we're now getting the desired behavior. Apparently, this is the default behaviour on Arduino (Our old Arduino based firmware just calls `pinMode(PIN_XY, INPUT);` and later `digitalRead(PIN_XY)`, relying on the default as well, which apparently seems to be the equivalent of `Pull::Floating`).

If putting the pin into pull-up mode by default is a deliberate decision, where is stuff like this documented, and why does it differ between the Rust toolchain and the Arduino one?

As a side note, I was, in contrast, pleasantly surprised by the ADC API in esp-idf-hal, where there's no default attenuation for the ADC, and instead one has to explicitly choose one of the available attenuation settings by using a const generic. Being forced to choose the right option instead of accidentally relying on an arbitrary default seems like the better API design choice and would have saved us several hours of debugging.

ivmarkov, 2024-01-26:
Here, the whole [saga](https://github.com/esp-rs/esp-idf-hal/pull/344) so far, for your education and entertainment :)

Seriously speaking, we were just discussing we plan to fix this to follow the "Arduino" model shortly after the new release of the `esp-idf-*` crates, which is happening as we speak.

ivmarkov, 2024-01-26:
[Also related](https://github.com/esp-rs/esp-idf-hal/issues/343) I think

ivmarkov, 2024-01-26:
BTW you might want to join the matrix chat: https://matrix.to/#/#esp-rs:matrix.org

jobafr, 2024-01-26:
Thanks for the pointers! Between the three different issue threads and the matrix channels, what is the preferred communication channel for this?

noahbliss, 2024-01-26:
> Thanks for the pointers! Between the three different issue threads and the matrix channels, what is the preferred communication channel for this?

Probably the issue here: https://github.com/esp-rs/esp-idf-hal/issues/343

ivmarkov, 2024-01-26:
Yes either this or the PR link.

Vollbrecht, 2024-05-25:
We changed the default behavior on transition and dropping, and its included in the latest point-release. I hope that solves the issue. In hardware terms the default now means every gpio register is set to its 0 value on drop and transition and in the transition case gets set to whatever the new state is. Here is the [code](https://github.com/esp-rs/esp-idf-hal/blob/2ae14905c67ab9b463a88bb1bba3849eb450b658/src/gpio.rs#L1444)

Closing it, feel free to add stuff and reopen it if you are not satisfied.

closed 2024-05-25 by Vollbrecht