esp-rs/esp-idf #373: Ubuntu 22.04.3 LTS Cross-Compilation for the target = "riscv32imac-esp-espidf" (IDFGH-11878)

blue-freedom-technologies, 2024-01-28:
Hello,

I am using the [ESP-IDF Toolchain](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/get-started/linux-macos-setup.html#standard-toolchain-setup-for-linux-and-macos) to compile the [test-app](https://github.com/blue-freedom-technologies/test-app).

```rust
[build]
target = "riscv32imac-esp-espidf"

[target.riscv32imac-esp-espidf]
linker = "/home/silveira/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/riscv32-esp-elf/bin/ld"
rustflags = ["--cfg", "espidf_time64", "-C", "target-feature=+crt-static"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c6"
ESP_IDF_VERSION = "v5.1.1"
force_ldproxy = "false"
TARGET_CC = "/home/silveira/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-cc"
```

But I am getting this error:

unrecognised emulation mode: arch=rv32imac_zicsr_zifencei
(Supported emulations: elf32lriscv elf64lriscv elf32briscv elf64briscv)

**if anyone have any suggestion for this [error](https://github.com/espressif/esp-idf/issues/12964) please comment.**

Cheers,

Vollbrecht, 2024-01-28:
Is there any specific reason why you manual try to set things like force_ldproxy and TARGET_CC etc? Please use the setup shown in the rust esp-idf-template. If you deviate from that you need to justify why first.

blue-freedom-technologies, 2024-01-28:
@Vollbrecht 

Yes, there is a reason why I am trying to do this.
Because I need to use this crate ["ring"](https://github.com/briansmith/ring) within my esp32-c6 app.

Cheers,

Vollbrecht, 2024-01-28:
By justify i mean you bring up technical reasons for that decision, not what you are thinking of archiving. So explaining why you do things the way you do.

I suggest you to look up how ldproxy works with [embuild](https://github.com/esp-rs/embuild) and how we integrating that into esp-idf-sys to build esp-idf.

blue-freedom-technologies, 2024-01-28:
Thank you for you message and advise.

"ring" exposes a Rust API and is written in a hybrid of Rust, C, and assembly language.

In order for this to work we need to set the TARGET_CC or equivalent to the C compiler for the target.

So, in order to make this work I need to setup this in this way, unless someone have a better approach of course.

Cheers,

closed 2024-02-06 by blue-freedom-technologies