esp-rs/esp-idf #377: Remove 4096 limit for dma transfer size

Kezii, 2024-02-10:
Why is there a limit on the transfer size for dma?

https://github.com/esp-rs/esp-idf-hal/blob/81675ccbf5d6868fedb3fa19a57d84ee5cfc06b0/src/spi.rs#L103

The esp32-s3 can supposedly go up to 32768 and previous models even higher, as referenced on https://github.com/espressif/esp-idf/issues/7804

Also, the documentation doesn't mention this number, it says, for max_transfer_sz

"Maximum transfer size, in bytes. Defaults to 4092 if 0 when DMA enabled, or to SOC_SPI_MAXIMUM_BUFFER_SIZE if DMA is disabled. "

* https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html

4092 is the default, and the number is wrong too

The 4096 limit is arbitrary, and should be removed

Vollbrecht, 2024-02-12:
You are correct that this limit is set arbitrary and may not make much sense. Though your assessments is about the 32768 is also not quite right as one has to think differently between actuall dma_descriptor lengths and what the C driver allows you to put a buffer into it. We had a discussion around it in the matrix channel some time ago. But in short, just removing it doesn't make the overall situation better.

What we would need is:
a) For every esp variant a verified dma transfer with the driver given arbitrary length buffers boxed on the heap for testing, and look what work and what doesn't work. 
b) check how the c spi_driver constructs the dma linked list out of the buffer we give him, and verify or deny any possible limit from the input buffer perspective.
c) Find out about possible penalty of non aligned buffers with respect to the dma descriptors itself. 

Based on that testing we could make a decision how to move forward correctly. So if you want to tackle the issue this are the first steps to dive into.

Kezii, 2024-02-12:
Isn't it wiser to just let the underlying esp-idf handle this? 

Right now it panics at runtime and it's not that much different than returning an error (what idf does when you remove the limit)

What you are saying is probably better but looks like a lot of work and a lot of maintenance, don't let perfect be the enemy of good

PR #379 (2024-02-18): Remove 4096 limit for SPI dma transfer size
fixes #377
closed 2024-02-18 by ivmarkov