esp-rs/esp-idf #378: Can't enable the `edge-executor` feature

weiying-chen, 2024-02-10:
I got this error:

```
the package `gpio` depends on `esp-idf-hal`, with features: `edge-executor` but `esp-idf-hal` does not have these features.
```
Note: `gpio` is my crate's name

According to this [link](https://github.com/esp-rs/esp-idf-hal/blob/master/Cargo.toml), you can enable `edge-executor` like this:

```
wake-from-isr = [] # Only enable if you plan to use the `edge-executor` crate
```

So I did that:

```
embassy = [
    "esp-idf-hal?/embassy-sync",
    "esp-idf-hal?/critical-section",
    "esp-idf-hal?/edge-executor", // edge-executor defined here
    "esp-idf-svc?/embassy-time-driver",
    "esp-idf-svc?/embassy-time-isr-queue",
]

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33", default-features = false }
esp-idf-hal = { version = "0.43", features = ["wake-from-isr"], optional = true, default-features = false } // wake-from-isr added here
esp-idf-svc = { version = "0.46", optional = true, default-features = false }
embedded-svc = { version = "0.25", optional = true, default-features = false }
anyhow = "=1.0.75"
```

But I'm still getting the same error.

ivmarkov, 2024-02-15:
* There is **no** feature `edge-executor` in `esp-idf-hal` anymore, as it is just not necessary
* Only enable `wake-from-isr` after you have the `edge-executor` running otherwise, as this is just an optimization
* Look at the *first* example on the [edge-executor website](https://github.com/ivmarkov/edge-executor). That's all you need to do to use `edge-executor` with `esp-idf-svcx/hal`.

closed 2024-02-15 by ivmarkov

weiying-chen, 2024-02-15:
Thanks for the reply!