esp-rs/esp-idf #384: Simple blink example for RGB Led on Esp32 C6

franxois, 2024-02-26:
Hi, 

I try to use the rmt_neopixel example on my esp32-c6 board with onboard rgb led but it doesn't work. My board is a nanoESP32-C6 (clone?). I only change gpio2 to 8 but the led don't blink.

On the other hand, the example on https://github.com/espressif/esp-idf/tree/release/v5.2/examples/get-started/blink/main works out of the box. 

I also tried with the blinky example, I notice a diff with the gpio8 init in the monitor log : 
working :         `gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
non working : `gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0`

How to achieve the same init in the blinky.rs code ? Is there another thing I should change to make the led blink in rust ? 

I choosed to not reopen the similar issue https://github.com/esp-rs/esp-idf-hal/issues/249

eteq, 2024-05-25:
I had the same problem as @franxois here... but after building with a version that has #402 in it, the problem went away!  So I'm pretty sure this can be closed now that #402 is in.

eteq, 2024-05-25:
(Oh, although I *might* be misunderstanding the OP here: I was trying to use the RMT on the RGB LED.  The "blink" example is not relevant to that because blink is only for single-color LEDs  that are tied directly to the GPIO.  But @franxois, if you indeed meant you're trying to get a ws2812-style RGB-LED working, then it makes sense that rmt_neopixel was broken, but now has nbeen fixed due to #402)

Vollbrecht, 2024-05-25:
Both classical blinky and RGB Leds should work on any boards with the current master branch. If its still doesn't work for OP please reopen it with a short code snippet of what you are doing.

closed 2024-05-25 by Vollbrecht

franxois, 2024-05-28:
Hello, thx for the update.

I confirm that the rmt_neopixel example works now on my board