esp-rs/esp-idf #388: Support async in i2c Driver by using new ESP-IDF V5.2 driver impl

teamplayer3, 2024-03-06:
[i2c Driver Docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.2/esp32/api-reference/peripherals/i2c.html)

The driver APIs have been clearly divided into master and slave, with all calls now separated into `i2c_master_*` and `i2c_slave_*`. This provides a higher-level interface for the i2c driver. To support async operations in master mode, callbacks can be set in `transmit`, `transmit_receive`, and `receive` to signal when the operation is complete. An event callback can also be registered with `register_event_callbacks` to receive events such as `I2C_EVENT_ALIVE`, `I2C_EVENT_DONE`, and `I2C_EVENT_NACK`. To initiate a receive job call `slave_receive` while in slave mode and to wait for any receive data a callback can be registered with `register_event_callbacks`.

## Changes to current i2c impl

- Add `AsyncI2cDriver`
- Add `AsyncI2cSlaveDriver`
- add cfg guards for current esp calls `major == 4 || version == 5.1`
- add cfg guards to new calls `major != 4 && version != 5.1`

kaaax0815, 2024-12-20:
Whats the status on this

darconeous, 2025-09-20:
Is anyone still working on this?