esp-rs/esp-idf #389: rmt_neopixel no longer works

thedevleon, 2024-03-06:
We used the [rmt_neopixel](https://github.com/esp-rs/esp-idf-hal/blob/master/examples/rmt_neopixel.rs) sample before to control an WS2812 RGB LED, however, it seems to have stopped in recent versions of esp-idf-hal / esp-idf-svc.

To reproduce, I have created a fresh project via `cargo generate esp-rs/esp-idf-template cargo` that targets the esp32c6, and copy-pasted the [rmt_neopixel](https://github.com/esp-rs/esp-idf-hal/blob/master/examples/rmt_neopixel.rs) sample.
I simply adjusted the `esp_idf_hal` imports to `esp_idf_svc::hal`, and the GPIO pin to suit our custom PCB.  

I get the following output, and the LED does not light up. 
```
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (56) boot.esp32c6: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 003f0000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=49f00h (302848) map
I (173) esp_image: segment 1: paddr=00059f28 vaddr=40800000 size=060f0h ( 24816) load
I (180) esp_image: segment 2: paddr=00060020 vaddr=42050020 size=20260h (131680) map
I (208) esp_image: segment 3: paddr=00080288 vaddr=408060f0 size=04d60h ( 19808) load
I (216) boot: Loaded app from partition at offset 0x10000
I (216) boot: Disabling RNG early entropy source...
I (228) cpu_start: Unicore app
I (228) cpu_start: Pro cpu up.
W (237) clk: esp_perip_clk_init() has not been implemented yet
I (244) cpu_start: Pro cpu start user code
I (244) cpu_start: cpu freq: 160000000 Hz
I (244) cpu_start: Application information:
I (247) cpu_start: Project name:     libespidf
I (252) cpu_start: App version:      1
I (256) cpu_start: Compile time:     Mar  6 2024 15:42:59
I (262) cpu_start: ELF file SHA256:  0000000000000000...
I (268) cpu_start: ESP-IDF:          v5.1.3
I (273) cpu_start: Min chip rev:     v0.0
I (278) cpu_start: Max chip rev:     v0.99 
I (283) cpu_start: Chip rev:         v0.0
I (288) heap_init: Initializing. RAM available for dynamic allocation:
I (295) heap_init: At 4080BE80 len 00070790 (449 KiB): D/IRAM
I (301) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (308) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (315) spi_flash: detected chip: generic
I (319) spi_flash: flash io: dio
W (323) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (333) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (342) sleep: Configure to isolate all GPIO pins in sleep state
I (349) sleep: Enable automatic switching of GPIO sleep configuration
I (356) coexist: coex firmware version: 77cd7f8
I (361) coexist: coexist rom version 5b8dcfa
I (366) app_start: Starting scheduler on CPU0
I (371) main_task: Started on CPU0
I (371) main_task: Calling app_main()
```

Please see the following repo to reproduce the issue, and make sure to adjust the GPIO to match the devkit you have:
https://github.com/thedevleon/esp-idf-led

I can rule out a defective LED or wrong GPIO, since a no_std sample based on the [esp-template](https://github.com/esp-rs/esp-template) and the [hello_rgp](https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/hello_rgb.rs) sample works just fine.

Vollbrecht, 2024-03-06:
Yeah i can at least confirm the problem. For comparison i run the rmt_musical_buzzer example and it still produces the output correctly.

adrianpanicek, 2024-03-08:
It works on `v0.42.5` some changes since must have broken it

thedevleon, 2024-03-12:
> It works on `v0.42.5` some changes since must have broken it
Can confirm, downgrading to `v0.42.5` and `ESP_IDF_VERSION = "v5.1.1"` makes the LED light up again.

PR #402 (2024-04-08): Fix FixedLengthSignal::set
https://github.com/esp-rs/esp-idf-hal/commit/5657073293b8995c11cdaf175c82d54d14dc8050 has broken the RMT driver. `Symbol(*item)` creates a copy that is never assigned to anything, making the effects of `update` useless.

Fixes #389
closed 2024-04-08 by Vollbrecht