esp-rs/esp-idf #392: Error: dangerous relocation: call8: call target out of range:(xtensa-esp32s3-espidf)

blue-freedom-technologies, 2024-03-11:
```text
= note: [ldproxy] Running ldproxy
          Error: Linker /esp32-s3/prototype/mcu-app/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc failed: exit status: 1
STDERR OUTPUT:
         /esp32-s3/prototype/mcu-app/target/xtensa-esp32s3-espidf/release/deps/libring-bdfba0d49cd2e9ff.rlib(d1a56b8165ef788f-curve25519.o): in function `fe_mul_impl':
         /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.ge_madd+0xf6): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
         /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.ge_madd+0x102): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
         /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.ge_madd+0x10d): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
         /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.x25519_ge_scalarmult_small_precomp$constprop$0+0x956): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
(...)
```
.cargo.toml
```text
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor"
#rustflags = [ "--cfg",  "espidf_time64"]
#rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries", "--cfg", "mio_unsupported_force_poll_poll","--cfg", "mlongcalls"]
rustflags = ["--cfg", "espidf_time64","--cfg", "mio_unsupported_force_poll_poll","--cfg", "mlongcalls","-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
ESP_IDF_VERSION = "v5.1.2"
TARGET_CC = "/home/silveira/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-cc"
```

In theory this happens when the binary blobs used is built without -mlongcalls. But in this case the  mlongcalls flag is being used.

Nevertheless, I am a bit lost with this error, any suggestion would be welcome.

Cheers,

MabezDev, 2024-03-11:
`"--cfg", "mlongcalls"` is doing nothing, remove it.

IIRC, mlongcalls isn't implemented in LLVM - but there is mtext-section-literals which you can try.

The way to pass args to llvm is

```
"-C", "llvm-args=-mtext-section-literals"
```

blue-freedom-technologies, 2024-03-11:
Thank you @MabezDev  but I have got the same error.

By the way you are right,  "-mlongcalls" is part of ESP-IDF standard flags. This flag is set in the toolchain file so it should propagate to all C source files even when using IDF as a library. 

```text
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor"
rustflags = ["--cfg", "espidf_time64","--cfg", "mio_unsupported_force_poll_poll","-C", "default-linker-libraries","-C", "llvm-args=-mtext-section-literals"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
ESP_IDF_VERSION = "v5.1.2"
TARGET_CC = "/home/silveira/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-cc"
```

```text
= note: [ldproxy] Running ldproxy
          Error: Linker /esp32-s3/prototype/mcu-app/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /esp32-s3/prototype/mcu-app/target/xtensa-esp32s3-espidf/release/deps/libring-bdfba0d49cd2e9ff.rlib(d1a56b8165ef788f-curve25519.o): in function `fe_mul_impl':
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.ge_madd+0xf6): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.ge_madd+0x102): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.ge_madd+0x10d): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.x25519_ge_scalarmult_small_precomp$constprop$0+0x956): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.x25519_ge_scalarmult_small_precomp$constprop$0+0x969): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.x25519_ge_scalarmult_small_precomp$constprop$0+0x15fe): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
          /.cargo/git/checkouts/ring-f7e9d4c00c5a74cb/a266f11/crypto/curve25519/curve25519.c:235:(.text.x25519_ge_scalarmult_small_precomp$constprop$0+0x1629): dangerous relocation: call8: call target out of range: .text.fiat_25519_carry_mul
(...)

```

This is so weird.

blue-freedom-technologies, 2024-03-13:
ok, with a bit of debugging I was able to fix this.

I need to add the "-mlongcalls" for the TARGET_CFLAGS used with the xtensa-esp32-elf-cc to compile the c code from the ring crate.

.cargo.toml
```text
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor"
rustflags = ["--cfg", "espidf_time64","--cfg", "mio_unsupported_force_poll_poll","-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
ESP_IDF_VERSION = "v5.1.2"
TARGET_CFLAGS = "-mlongcalls"
TARGET_CC = "/home/silveira/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-cc"
```

closed 2024-03-13 by blue-freedom-technologies