esp-rs/esp-idf #395: SpiBusDriver is not Send

leonardodanelutti, 2024-03-19:
Is there a reason why Send is implemented for `SpiDeviceDriver` and not for a `SpiBusDriver`? From the documentation there seems to be no difference between a `SpiBusDriver` without cs and a `SpiBusDriver`.

ivmarkov, 2024-04-30:
This should be fixed. PRs welco.e!

Backgroynd: All drivers should be `Send` but currently few (if any) are `Sync`. We can actually also do the latter too for most drivers because it turns out most esp idf APIs are thread safe.

But usually folks reaching out to `Sync` often means they might be doing something suboptimal, as in - say - trying to use the default tokio work-stealing executor, which is not ideal on embedded.

Vollbrecht, 2024-06-21:
Still a workaround but should do it for most people is just sending the peripheral or the the wrapped `SpiDriver` as it is `Send`