esp-rs/esp-idf #398: Problem with CHANGELOG.md in v0.43.0 release: Unable to find valid link for esp-ulp-riscv-hal crate.

deepak1824, 2024-04-01:
I've been trying to use the latest release v0.43.1 for my ULP application. However, there seems to be an issue with the CHANGELOG.md file. The change logs for the release tag v0.43.0 mention a breaking change related to the removal of the `riscv-ulp-hal` feature. It suggests using the `esp-ulp-riscv-hal` crate instead. However, when I try to access the crate through the provided link, I receive a 404 error.

[image]

Is there a valid link or crate for the `esp-ulp-riscv-hal` or an alternative method to use the `esp-idf-hal` crate in my ULP application?

Additionally, if there are any examples available showcasing how to use ULP with Rust, it would be greatly appreciated. This would help in understanding how to integrate ULP functionality into Rust applications effectively with the updated code.

### Additional Information
- Operating System: macOS
- Espressif Chip: esp32s3
- Rust Version: 1.76.0

Vollbrecht, 2024-04-02:
The ulp repository is [here](https://github.com/esp-rs/esp-hal/tree/main/esp-lp-hal). What you are basically doing is building this specifc riscv-unknown-unknown target as a standalone elf, and then loading it into your normal main like you did previously. For how to program the ulp itself you should find minimal examples there.

The methods for loading and running the ulp are still found inside the ulp.rs of this project, though not sure how tested they are now with the "new" approach. There is no example for it but nothing changed here from before.

deepak1824, 2024-04-02:
After doing some internal digging, we figured out where the rust crate for the low-power RISC-V coprocessors is present https://github.com/esp-rs/esp-hal/tree/main/esp-lp-hal. It appears that multiple times refactoring was done to the Low processor code. Following are the few commits where it helped us track it down.
https://github.com/esp-rs/esp-hal/tree/47821e6b3b6a6d811041856f020c42f0bf6083bc/ulp-riscv-hal
https://github.com/esp-rs/esp-hal/tree/c612fecfaef9f59473057d826d480785ec3fa513/esp-ulp-riscv-hal
https://github.com/esp-rs/esp-hal/pull/1115

However, the link is still incorrect in the change logs and would be beneficial if it got updated.