esp-rs/esp-idf #422: esp32 - Spi Drivers freeze app and triggers TG1WDT_SYS_RESET

Vollbrecht, 2024-05-22:
By esp32 -> classical esp32

Key observation is that it does work with gpio 15-18 for sclk,mosi,miso,cs and some minimal random other pin combinations, changing  one or two of them, but most combinations of pinnumbers will freeze the esp and trigger a reset after varying seconds.

Use default cargo generate template. 

Minimal repo:
```rust
use esp_idf_svc::hal::peripherals::Peripherals;
use esp_idf_svc::hal::spi::{SpiDeviceDriver, SpiDriverConfig, SPI2};

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    // Bind the log crate to the ESP Logging facilities
    let peripherals = Peripherals::take().unwrap();

    let sclk = peripherals.pins.gpio15;
    let miso = peripherals.pins.gpio12;
    let mosi = peripherals.pins.gpio11;
    let cs = peripherals.pins.gpio13;

    let driver_conf = SpiDriverConfig::new();
    let driver = SpiDeviceDriver::new_single::<SPI2>(
        peripherals.spi2,
        sclk,
        miso,
        Some(mosi),
        Some(cs),
        &driver_conf,
        &Default::default(),
    )
    .unwrap();

    loop {
        log::info!("Hello, world!");
        std::thread::sleep(std::time::Duration::from_secs(1));
    }
}

``

DaneSlattery, 2024-07-16:
Does this mean that SPI will not work at all when assigned to different pins?

What pins actually can be used?

Those given in the `esp-idf-hal/examples/loopback`?
- CLK on GPIO15
- MISO on GPIO17
- MOSI on GPIO16
- CS1 on GPIO18

(PS, the example file comments are lying).

[rust-esp32-demo](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L277) uses these pins:
- CLK GPIO13
- MISO on GPIO12
- MOSI on GPIO26
- CS on GPIO14

The [ESP-IDF docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux) suggests the use of these pins for SPI2
- CLK on GPIO14
- MISO on GPIO12
- MOSI on GPIO13
- CS1 on GPIO15

And finally, [esp-idf C examples](https://github.com/espressif/esp-idf/blob/v5.2.2/examples/ethernet/README.md#common-pin-assignments) suggest these pins:
- CLK on GPIO14
- MISO on GPIO12
- MOSI on GPIO13
- CS on GPIO15

I had intended to connect to a KSZ8851SNL and an SD Card via SPI, and I don't want to get my pin assignments wrong when I submit to the PCB manufacturer

Vollbrecht, 2024-07-17:
The recommendation is to use this special gpio's because of potential minimal latency gains, since this gpio's dont need to go through the gpio matrix. Though they claim that any gpio can be used for the SPI2 device. quote: "However, the signals can also be routed to any other available pins using the less direct GPIO matrix." And that should be certainly true. I tried to debug it a bit but for some unknown reason my esp hard crashes and it's hard to debug the issue currently. I only currently now that it happens somewhere in the initiation phase inside the gpio matrix code path.

DaneSlattery, 2024-07-17:
I guess my question is, which special GPIOs actually do work? And which devices are affected?